cnlohr/ch32fun #4: Two stage boot loader

jonsmirl, 2023-02-23:
Have you looked at a two stage boot loader process which would open this device up to the class of people who won't use a WCHLink? Companies making hobbyist devices would flash in the first stage during production. That first stage would then allow the hobbyist to update the flash using the UART, eliminating the need for the WCHLink. I don't know how feasible this is given the tiny amount of flash/ram.

cnlohr, 2023-02-24:
If anything you should go the opposite direction.  Flashing over UART would not be an ideal way to do things.  Instead it would be better to just flash over the one-wire interface because it is (1) unbrickable, and (2) seems to have versatility to do debugging, etc.  (3) It also works under virtually all clocking schemes. (4) Flashing over UART loses you the UART and (5) UART flashing requires 2 pins instaed of one.

I am not a super expert here, but I very very strongly feel hobbyists who make flashers should just use the native protocol.  I RE'd it in the "attic" doc in this repo, and, also, WCH has released additional information about the one-wire-flash protocol.

It looks pretty easy to write flasher code to just use a GPIO to flash the CH32V003.  I plan to write a platform-independent flasher, much like my pi_tpi project where the entire flasher is written in a way to be portable to any platform.

https://github.com/cnlohr/pi_tpi/blob/master/gpio_tpi.c

cnlohr, 2023-02-28:
ESP32S2, or ESP32S3.  The ESP32 doesn't natively have USB.

jonsmirl, 2023-02-28:
Can you make it a library? I was planning to send the file via wifi.

cnlohr, 2023-02-28:
One reason I like the ESP32S2 is because it supports both wifi and USB.  I am definitely leaning toward making an ESP32S2 programmer.

cnlohr, 2023-03-01:
I have it all working with the ESP32-S2 while it remains connected to USB.  https://github.com/cnlohr/esp32s2-cookbook/blob/master/idf_sandbox/tools/ch32v003_swio_interface_test/ch32v003_swio.h 

Just by connecting a 10k ohm resistor to Vcc and connecting the pin to the SWIO pin on the CH32V003.

I plan to further develop this to add functions like flash_frimware, etc.  Right now, it's basically just a header file and a little bit of wrapping (see the sandbox file in the folder).  But I will likely make this into a real "IDF" project.

cnlohr, 2023-06-21:
In general this issue has been largely bypassed in relevancy.  There are lots of options available and people can work on whichever they want.  We also have rv003usb now as well.

closed 2023-06-21 by cnlohr