cnlohr/ch32fun #12: Fix warnings

maxgerhardt, 2023-03-04:
Compiler goes Mi-mi-mi with `-Wall` on certain code files, this could lead to bugs if someone expects standard-compliant libc functions.
```
src\embedlibc.c: In function '__puts_uart':
src\embedlibc.c:18:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
In file included from src/ch32v00x.h:2396,
                 from src\startup_ch32v003.c:8:
src/ch32v00x_conf.h:1784:13: warning: 'DelaySysTick' defined but not used [-Wunused-function]
 static void DelaySysTick( uint32_t n )
             ^~~~~~~~~~~~
c:\users\max\.platformio\packages\toolchain-riscv\riscv-none-embed\include\string.h: In function 'memset':
src\embedlibc.c:108:1: warning: control reaches end of non-void function [-Wreturn-type]
 void *memset(void *dest, int c, size_t n) { unsigned char *s = dest; for (; n; n--, s++) *s = c; }
 ^~~~
c:\users\max\.platformio\packages\toolchain-riscv\riscv-none-embed\include\string.h: In function 'strcpy':
src\embedlibc.c:109:1: warning: control reaches end of non-void function [-Wreturn-type]
 char *strcpy(char *d, const char *s) { for (; (*d=*s); s++, d++); }
 ^~~~
c:\users\max\.platformio\packages\toolchain-riscv\riscv-none-embed\include\string.h: In function 'strncpy':
src\embedlibc.c:110:1: warning: control reaches end of non-void function [-Wreturn-type]
 char *strncpy(char *d, const char *s, size_t n) { for (; n && (*d=*s); n--, s++, d++); }
 ^~~~
```

cnlohr, 2023-03-05:
All good catches.  I will fix 'em tomorrow.

cnlohr, 2023-03-06:
I meant if you switch `ch32v003fun/ch32v003fun.c` / `.h`

I can also switch the WS2812 project tonight. 

I guess I can used `__attribute__((used))` to nix the warning, and also prevent inclusion into the executable.  

I am always leary to accept CI stuff since I very rarely work with it myself.  Does this project interest you in a longer-scale time-frame?  That if I have questions or issues with the CI stuff you would still probably take a look?

maxgerhardt, 2023-03-06:
Sure'd have a look at CI if anything breaks. I'm the current and only maintainer of platform-ch32v after all. 

Though I think you emant to reply in the PR? Doesn't matter either way.

cnlohr, 2023-03-20:
Should we close this issue?

maxgerhardt, 2023-03-20:
Well...
```
examples\debugprintfdemo\debugprintfdemo.c:33:14: warning: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
   printf( "+%d\n", count++ );
             ~^     ~~~~~~~
             %ld
examples\debugprintfdemo\debugprintfdemo.c:36:14: warning: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
   printf( "-%d\n", count++ );
             ~^     ~~~~~~~
             %ld
ch32v003fun\ch32v003fun.c: In function '_write':
ch32v003fun\ch32v003fun.c:900:12: warning: unused variable 'd' [-Wunused-variable]
   uint32_t d;
            ^
examples\sandbox\sandbox.c: In function 'SRAMCode':
examples\sandbox\sandbox.c:24:1: warning: 'noreturn' function does return
 }
 ^
 
examples\self_modify_code\self_modify_code.c: In function 'main':
examples\self_modify_code\self_modify_code.c:94:23: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
    printf( "%03x = %08x\n", i, rv );
                    ~~~^        ~~
                    %08lx
examples\uartdemo\uartdemo.c:32:20: warning: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
   printf( "Count: %d\n", count++ );
                   ~^     ~~~~~~~
                   %ld
examples\ws2812bdemo\ws2812bdemo.c:27:30: warning: suggest parentheses around '+' in operand of '&' [-Wparentheses]
  uint32_t fire = huetable[(rs+190&0xff)] | (huetable[(rs+30&0xff)]<<8) | (huetable[(rs+0)]<<16);
                            ~~^~~~
examples\ws2812bdemo\ws2812bdemo.c:27:57: warning: suggest parentheses around '+' in operand of '&' [-Wparentheses]
  uint32_t fire = huetable[(rs+190&0xff)] | (huetable[(rs+30&0xff)]<<8) | (huetable[(rs+0)]<<16);
                                                       ~~^~~
In file included from examples\ws2812bdemo\ws2812bdemo.c:13:
At top level:
examples\ws2812bdemo\color_utilities.h:12:17: warning: 'EHSVtoHEX' defined but not used [-Wunused-function]
 static uint32_t EHSVtoHEX( uint8_t hue, uint8_t sat, uint8_t val )
                 ^~~~~~~~~
```

Maybe it also helps to add `-Wall` into the makefiles to make these warnings more apparent.

cnlohr, 2023-03-23:
Should be fixed in 1d26dfa

closed 2023-03-23 by cnlohr