cnlohr/ch32fun #14: Installation instructions are ambiguous.

murray-macdonald, 2023-03-05:
The installation instructions are ambiguous.   It would be helpful if the paths and folder references were explicit, rather than vague.  They could even include links to the correct references.  As they exist, they assume many things that are not evident

cnlohr, 2023-03-06:
Things are still a little vague, indeed.

Right now, all of it works with WSL in Windows, or any mainline version of Linux that has a copy of a RISC-V GCC. 

But there is not currently a downloadable-from-here copy of the RISC-V compiler.

All building is done by `cd`'ing into the folder with the project or demo you want to run and typing `make`

Any idea how I could word that better for the readme?

cnlohr, 2023-03-06:
Ok, I've updated the instructions for Windows, directing people to use https://gnutoolchains.com/risc-v/

closed 2023-03-06 by cnlohr