cnlohr/ch32fun #54: makefile bricked on linux??

recallmenot, 2023-04-27:
Hi,
I'd like to report a bug that I've got no idea how to solve.
I am using two OSes: Manjaro and Kubuntu 22.10
About a week ago I was using Kubuntu to upload using the WCH-Link E.
Then I spent a week tryping to wrap my head around SPI, now "sudo make" builds an .elf but doesn't upload. I'm using the latest "unified makefile".
With blink I get elf and bin but no upload.
On manjaro I can remember make always calling minichlink to upload.
On Kubuntu everything just worked, no issues, and I haven't run updates since.
Strange thing is: when I manually call minichlink with the right bin, it gets flashed like it used to.
But sudo make just successfully builds, no errors, no "programmer-not-found".
Seems like it just doesn't call minchlink and in some examples doesn't generate a .bin file, just the elf.
Maybe it's fine for yall on Windows?
After restoring blink to the "old" style of makefile
https://github.com/cnlohr/ch32v003fun/commit/d71a0500c2145f920a345081392367e387298030
that one just works again!
I even did a fresh wipe and clone from master an hour ago!

emeb, 2023-04-27:
The make targets were changed recently so that the default with no target specified now just builds the binary. When you want to program you must now use the "flash" target. In other words:

make flash

in order to program.

recallmenot, 2023-04-27:
ah that works, thank you @emeb!

closed 2023-04-27 by recallmenot

cnlohr, 2023-04-27:
Perhaps we should switch that back to default.

cnlohr, 2023-04-27:
https://github.com/cnlohr/ch32v003fun/pull/56