cnlohr/ch32fun #65: Add fence instruction for memory rewriting.

cnlohr, 2023-05-03:


cnlohr, 2023-05-03:
(Memory writing of code)

dwillmore, 2023-05-03:
Is this related to debugging?  Because there's no caches on this chip, so the only things you could be fighting with memory fences would be prefetch queues, etc.

cnlohr, 2023-05-04:
I have not yet had an issue when debugging, but when I modified code in RAM, it was required. There is definitely _some_ kind of cache.

https://github.com/cnlohr/ch32v003fun/blob/master/examples/self_modify_code/self_modify_code.c#L50

Without this line, things don't work right.

jnk0le, 2023-05-04:
Fetch of next instruction happens before the store is commited, most likely.

Shouldn't it be `fence.i`, btw?

cnlohr, 2023-05-05:
I will have to experiment to see what is required.

cnlohr, 2023-05-11:
It looks like this has not been and likely will not be an issue. I was worried something may be janky with debugging.

closed 2023-05-11 by cnlohr