cnlohr/ch32fun #72: Link-E Support for non-ch32v003 chips

cnlohr, 2023-05-06:
@bmx ran into an issue with this - trying to use it on CH32V307 boards. 

I did not know it even could run on a 307.  I don't mind trying to support this some moving forwards, but I think I need a way to identify if it's a 307 or a 003?  Do you know much about this?  Are the debug protocols similar?

https://github.com/cnlohr/ch32v003fun/commit/6f896b6fe749481f8a6d9d96e0b7730545355797

Side-note: If we can discuss this on the Discord for this project that would be helpful.

cnlohr, 2023-05-06:
I have modified it so if it detects a non-003 it tries to go into the old flashing mode.  I "think" this should fix you, BMX. 249f5d2

bmx, 2023-05-08:
```Found WCH LinkE
Part Type (A): 0x0617 (This is the capacity code, in KB)
Part UUID    : 00-26-10-00-00-ba-06-71
PFlags       : 23-48-76-05
Part Type (B): e4-59-a8-e9
Interface Setup
Fault writing memory (DMABSTRACTS = 08000302) (Execption executing Abstract Command) DMSTATUS: 00030382
Fault reading device```

using this commit.

We need to disambiguate the use of `ReadBinaryBlob` which is meant to read-many-bytes on the target, not dealing with the proprietary blob.

cnlohr, 2023-05-08:
I believe this is now resolved.  Should work with the 003 and 307 among others.

closed 2023-05-08 by cnlohr