cnlohr/ch32fun #89: Error 'unrecognized opcode csrrs' when compiling self_modify_code example with gcc 12.x

cw2, 2023-05-10:
Unfortunately, due to a breaking change in RISC-V ISA 20191213 (https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/aE1ZeHHCYf4)

>... the csr read/write (csrr*/csrw*) instructions and fence.i instruction has separated from the `I` extension, become two standalone extensions: Zicsr and Zifencei;...

gcc 12.x now requires `-march=rv32ec_zicsr` in order to compile `csrr*` instructions. Although it is an easy fix when supporting just the latest gcc 12.x risc-v toolchains, the question is how to do it while keeping backward compatibility with older gcc toolchains? E.g. a conditional parameter assignment in makefile based on gcc version?

cnlohr, 2023-05-10:
Can this be resolved by adding a `.option zicsr` in front of the code?

cw2, 2023-05-10:
Yes! I have missed it is actually already used elsewhere...
```C
  asm volatile(
...
#if __GNUC__ > 10
  ".option arch, +zicsr\n"
#endif
```
Nice 👏

cw2, 2023-05-10:
Arch option added in https://github.com/cnlohr/ch32v003fun/pull/88/commits/fc598d183d0c8f9aa72d568d3a7167e8d493a8da

cnlohr, 2023-05-10:
Thanks. Good catch.  If you can, address the comment on your PR and then we can merge this in!

commit 186d9dcf (2023-05-14): Added zicsr arch option. Closes #89
commit d18ad76c (2023-05-14): Merge pull request #97 from cw2/option-zicsr

Added zicsr arch option. Closes #89
closed 2023-05-14 by cnlohr