cnlohr/ch32fun #98: esp32s2 programmer failed after second run on j4m6

phantomxe, 2023-05-17:
I got some error message at try to program ch32v003's 8-pin package. It works on fresh chip but crashes when i try to upload new code.  

```
Found ESP32S2 Programmer
Error: Setup chip failed. Got code ffffffff
Could not setup interface.
``` 

I also tried minichlink -u mode. 

```
Found ESP32S2 Programmer
Entering Unbrick Mode
Connection starting
/0/ffffffff
/0/ffffffff
...
lots of ffffffff
...
ioctl (GFEATURE): Connection timed out
Error: Got error -1 when sending hid feature report.
Error: Flash is not unlocked (CTLR = ffeffdff)
```

dwillmore, 2023-05-17:
Have you tried using the "-u" option of minichlink?  You need to let the programmer control power to the target if you aren't already.

phantomxe, 2023-05-17:
I added power control circuit with pair of mosfets to GPIO11 and tried `minichlink -u` again.

```
Found ESP32S2 Programmer
Entering Unbrick Mode
Connection starting
/0/ffffffff
/0/ffffffff
/0/ffffffff
/0/ffffffff
/0/ffffffff
/0/ffffffff
/0/004c0c82
Flash unlocked
Whole-chip erase
Whole Chip Erase Code: 0
```

Seems to work. But when i try to program not working.

```
Found ESP32S2 Programmer
Error: Setup chip failed. Got code ffffffff
Could not setup interface.
```

cnlohr, 2023-05-18:
Two thoughts:
I have seen situations where the whole-chip erase was unsuccessful - can you try `minichlink -ua`  If that works, let me know and I can make that default behavior.

phantomxe, 2023-05-18:
`minichlink -ua` worked! I can reprogram chip without any issues now.

[image]

closed 2023-05-18 by phantomxe