cnlohr/ch32fun #99: CH32V307 Variant

wrcrooks, 2023-05-18:
I know you're probably busy working on the rv003usb project right now, but any chance you can create a "ch32v307fun.c" file based off the CH32V307_startup files that OpenWCH has [here](https://github.com/openwch/ch32v307)?

The eval boards have a lot more IO and even 100M ethernet. I followed your lead on adapting the header files provided by OpenWCH but have gotten hung up on some of the Assembly instructions that would be used by the linker. I can provide my work so far if that'd help put everything in context but your ch32v003fun repo is by far smoother than MounRiver Studio.

-wrcrooks

cnlohr, 2023-05-20:
I am very interested in this. There are some other folks as well.  I am still exploring the ch32v003, and I don't have the bandwidth to work on the 307 as well.  But it would be amazing for someone else to work on it. 

I am happy for there to be a ch32v307fun folder alongside the 003 that contains a similar structure and feature set if you are up for working on it!

TommyMurphyTM1234, 2024-06-18:
Could this issue perhaps be closed now? 

- https://github.com/cnlohr/ch32v003fun?tab=readme-ov-file#other-family-support
- https://github.com/cnlohr/ch32v003fun/tree/master/examples_v30x/blink
- https://github.com/cnlohr/ch32v003fun/blob/ef18357d9bc518466f1abaab7dbd268abee614a9/ch32v003fun/ch32v003fun.mk#L126-L127

cnlohr, 2024-06-19:
It does look like it's done.

closed 2024-06-19 by cnlohr