cnlohr/ch32fun #127: Remove parts of Readme in favor of wiki

cnlohr, 2023-05-27:


mengstr, 2023-05-28:
Maybe mentioning that the 20 ns resolution @48MHz is really closer to 21 or add a few decimals to it could be a good thing? My mind went straight to "_ohhh, nice even number, that seems really useful!_" before realising that it actually is 20.83333...

cnlohr, 2023-05-30:
Where is that mention?

mengstr, 2023-05-31:
It currently says that the resolution is 20ns, but I think it would be nice to show a few decimals in order to not have people believe that it's exactly 20ns (as it would be if the chip was run at 50MHz). I know the formulas and all is there, but my eyes was drawn to the nice numbers. ;)

https://github.com/cnlohr/ch32v003fun/wiki/Time#time-by-systick

[image]

cnlohr, 2023-05-31:
fixed

closed 2023-06-21 by cnlohr