cnlohr/ch32fun #142: Example request - alternate function mapping / pin mux

prosper00, 2023-05-30:
Some pins on the 8-pin package have multiple functions, as well as multiple assignments.

Port D4 and port D5 are both listed as available on pin 8, for example, as well as having default and remapped functions available for each. I would find it helpful to have an example showing how to select between GPIOs on that pin, as well as between the alternate functions.

[image]

GerardoNevarez, 2023-05-30:
All default functions for these "muxed" pins are available by default. When you set, say, pin8 as UTX, you can no longer use it for programming the IC.

See the various examples that are already present, the short version is that _both_ default and alternative functions have to be explicitly set, by changing configuration registers for each port

dwillmore, 2023-05-30:
There isn't another set of 'alternate function' registers for these
combined pins.  We haven't verified this by decapsulating a chip, but it's
likely that both pads on the chip have been bonded to the one pad on the
lead frame of the package.

You could enable one of those 'pins' (from the perspective of the chip) as
well as the other 'pin' which share a package pin as long as they don't
both try to drive different signals.  If one was input only and the other
set to output, you could do that.  I'm not sure how useful it might be.

Another possibility is to increase the drive output of a package pin by
enabling both of the internal pins as output and driving them
simultaniously.  Experiments have yet to be done to gauge the possible
benefit of that.

prosper00, 2023-05-30:
Interesting. That was my reaction - 'what if I write a 'high' to PD4 and a 'low' to PD5?' Seems likely to be a smoke-emitting configuration.

If I enable a peripheral on PD4, will it disable all the peripherals on PD5 - or do I just have to be careful? 

in any case, my original question has been answered, and it's really quite simple. 

Thanks!

closed 2023-05-30 by prosper00

dwillmore, 2023-05-30:
The current untested theory is you have to be careful. :)  It very well may
emit smoke if mishandled. :)  Proceed with caution.  If you learn anything,
please come back here and start a discussion to share your knowledge.