cnlohr/ch32fun #143: printf needs to have tunable timeout

cnlohr, 2023-05-30:


prosper00, 2023-06-05:
Perhaps you could adopt a similar approach to Segger's RTT lib? It allows you to configure various buffer options, and modes (i.e. what to do if the buffers are full; blocking wait (though it doesn't have a timeout that I'm aware of - but I never use blocking mode), overwrite, skip, etc).

If you're not already familiar, RTT code is hosted here: https://github.com/adfernandes/segger-rtt/ - and Segger's wiki is here: https://www.codeinsideout.com/blog/stm32/segger-rtt/#how-rtt-works 

Come to think of it, RTT would probably work as-is on the CH32V003, if you're using a recent-ish OpenOCD that supports RTT (and of course, the CH32V003, and a debug probe)

thirstyone, 2023-11-01:
> Come to think of it, RTT would probably work as-is on the CH32V003, if you're using a recent-ish OpenOCD that supports RTT (and of course, the CH32V003, and a debug probe) 
any success with OpenOCD

monte-monte, 2025-11-25:
I think we have it now?