cnlohr/ch32fun #145: minichlink -t and -a don't behave as expected (reset through minichlink)

recallmenot, 2023-06-01:
When looking for a way to reset the ch32v003 through minichlink using Link-E, I've noticed, that:
`-t` should disable it (poweroff) so I can later bring it back with `-3`
 * power-off as expected
 * gets stuck at `Found WCH Link` for a couple of seconds
 * then errors out with `Error sending WCH command (on recv): 81 0d 01 ff `
 * power is returned automatically (wanted too do that manually in the next step)
 
 `-a` runs fine but doesn't halt execution (blink still continues), which I'd then want to reset with `-b`  

Am I using both wrong?  
Ideally, I'd like to have PD7 available for other stuff.  

Some related suggestions:
 * What do you think of the idea of having `reset` in the makefile so it's accessible from anywhere?
 * Maybe add an option that resets it but places it in a state expecting a flash, but without the erase-program cycle of `-u`? With the standby examples, I've often had to use `-u` to flash anything because it would be awake for such a short time it wouldn't communicate at all.

recallmenot, 2023-06-01:
from trying it just now, even `-ba` doesn't result in halted execution, blink still running

cnlohr, 2023-06-09:
I am hoping that the improved definitions for `-a` (do reboot into halt) and `-A` (truly pause) resolves some of these issues.  

This should prevent the need to use `-u`.  

I can look into why `-a` doesn't truly halt.

cnlohr, 2023-06-09:
Ok, looks like `-A` works correctly.

cnlohr, 2023-06-21:
I am closing now because I think?  It's been resolved?

closed 2023-06-21 by cnlohr