cnlohr/ch32fun #146: Prebuilt MacOS minichlink missing

mengstr, 2023-06-02:
https://github.com/cnlohr/ch32v003fun/wiki/Installation states under MacOS ```You can use the pre-compiled minichlink or go to minichlink dir and make it.```

But there is no prebuilt executable for MacOS in the repo.  Should there be one? It's kinda practical to have ready-made executables, but I guess most people will create them without any issues.

maxgerhardt, 2023-06-02:
I think it was there at some point but got deleted by a commit. But I can just add CI to build minichlink for every possible configuration and then link it.

maxgerhardt, 2023-06-02:
Can I ask you to download and try the right MacOS from https://github.com/Community-PIO-CH32V/ch32v003fun/actions/runs/5157203799 as test? 

Btw are .tar.gz or .zip files more popular for MacOs Downloads? You'd need to first unzip then untar (`tar xfv <file.tar.gz>`) the archive, then `./minichlink -h` as execution test.

mengstr, 2023-06-02:
No joy. I tried the MacOS 12 download (I'm actually on MacOS 13) and I got a zip that contained a tar.gz which ended up as a `minichlink: Mach-O 64-bit executable x86_64`.

After the usual complaints about unsigned code it ran after I approved it. But resulted in a missing library.

```
mats@Matss-MacBook-Pro Downloads % ./minichlink
dyld[69320]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <4EABE23C-602B-39A1-9247-81A79DB348EC> /Users/mats/Downloads/minichlink
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
zsh: abort      ./minichlink
```
Which is reasonable since I'm having a M1 CPU and it runs X86 code emulated in Rosetta. But it then can't use my installed M1 libusb. (At least I think it's like that ^__^)

When I compile it myself using the Makefile I get a M1/ARM executable `./minichlink:  Mach-O 64-bit executable arm64`that loads the M1 libusb

```mats@Matss-MacBook-Pro minichlink % otool -L minichlink
minichlink:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.100.3)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1971.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
```

It probably would work better if you can compile it for MacOS 13 / Ventura in CI.

Personally I always download a .tar.gz if given a choice between that and a zip. But that is probably only because I've been using linux on servers since the 90s.  But either is probably just fine.

maxgerhardt, 2023-06-02:
I've seen this trick here once

https://github.com/earlephilhower/arduino-pico/issues/703#issuecomment-1192974035

Can you try the same to try and install the x86_64 version of libusb-1.0?

Unfortunately I can't use Github provided runners for MacOS 13 or ARM M1 because [they aren't yet generally availbale](https://docs.github.com/de/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources),

mengstr, 2023-06-02:
I got it working but it's far from ideal. It's much simpler to have the user make the executable themselves.

`arch -x86_64 brew install libusb` as mentioned in the linked  issue didn't work. It tried to just install the M1 library again.

But following the steps at https://medium.com/mkdir-awesome/how-to-install-x86-64-homebrew-packages-on-apple-m1-macbook-54ba295230f did install the x86 libs
```
curl -L https://github.com/Homebrew/brew/tarball/master | tar xz --
mv Homebrew-brew-f7d86ff homebrew
sudo mv homebrew /usr/local/homebrew\n
export PATH=$HOME/bin:/usr/local/bin:$PATH
alias axbrew=arch -x86_64 /usr/local/homebrew/bin/brew
axbrew install libusb
```

But the minichlink executable is looking for the library in the following locations:
```
 /usr/local/opt/libusb/lib/libusb-1.0.0.dylib 
/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib 
/usr/local/opt/libusb/lib/libusb-1.0.0.dylib 
/usr/local/lib/libusb-1.0.0.dylib
/usr/lib/libusb-1.0.0.dylib
```

and the x86 brew puts it elsewhere
```
find /usr/local | grep libusb   
/usr/local/homebrew/include/libusb-1.0
/usr/local/homebrew/var/homebrew/linked/libusb
/usr/local/homebrew/Library/Taps/homebrew/homebrew-core/Formula/libusb.rb
/usr/local/homebrew/Library/Taps/homebrew/homebrew-core/Formula/libusbmuxd.rb
/usr/local/homebrew/Library/Taps/homebrew/homebrew-core/Formula/libusb-compat.rb
/usr/local/homebrew/lib/pkgconfig/libusb-1.0.pc
/usr/local/homebrew/lib/libusb-1.0.dylib
/usr/local/homebrew/lib/libusb-1.0.0.dylib
/usr/local/homebrew/lib/libusb-1.0.a
/usr/local/homebrew/opt/libusb
/usr/local/homebrew/Cellar/libusb
/usr/local/homebrew/Cellar/libusb/1.0.26
/usr/local/homebrew/Cellar/libusb/1.0.26/INSTALL_RECEIPT.json
/usr/local/homebrew/Cellar/libusb/1.0.26/.brew
/usr/local/homebrew/Cellar/libusb/1.0.26/.brew/libusb.rb
/usr/local/homebrew/Cellar/libusb/1.0.26/ChangeLog
/usr/local/homebrew/Cellar/libusb/1.0.26/AUTHORS
/usr/local/homebrew/Cellar/libusb/1.0.26/include
/usr/local/homebrew/Cellar/libusb/1.0.26/include/libusb-1.0
/usr/local/homebrew/Cellar/libusb/1.0.26/include/libusb-1.0/libusb.h
/usr/local/homebrew/Cellar/libusb/1.0.26/README
/usr/local/homebrew/Cellar/libusb/1.0.26/TODO
/usr/local/homebrew/Cellar/libusb/1.0.26/COPYING
/usr/local/homebrew/Cellar/libusb/1.0.26/NEWS
/usr/local/homebrew/Cellar/libusb/1.0.26/lib
/usr/local/homebrew/Cellar/libusb/1.0.26/lib/pkgconfig
/usr/local/homebrew/Cellar/libusb/1.0.26/lib/pkgconfig/libusb-1.0.pc
/usr/local/homebrew/Cellar/libusb/1.0.26/lib/libusb-1.0.dylib
/usr/local/homebrew/Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib
/usr/local/homebrew/Cellar/libusb/1.0.26/lib/libusb-1.0.a
/usr/local/homebrew/Cellar/libusb/1.0.26/share
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/sam3u_benchmark.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/ezusb.h
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/fxload.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/ezusb.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/xusb.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/dpfp.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/testlibusb.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/hotplugtest.c
/usr/local/homebrew/Cellar/libusb/1.0.26/share/libusb/examples/listdevs.c
/usr/local/homebrew/share/libusb
/usr/local/flutter/bin/cache/artifacts/usbmuxd/libusbmuxd-2.0.6.dylib
/usr/local/opt/libusb
/usr/local/opt/libusb/lib
/usr/local/opt/libusb/lib/libusb-1.0.0.dylib
```

Copying the libraty to a place where minichlink looks it then works
```
sudo mkdir -p /usr/local/opt/libusb/lib/
sudo cp /usr/local/homebrew/Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib /usr/local/opt/libusb/lib/

~/Downloads/minichlink -h                                                                         
Usage: minichlink [args]
 single-letter args may be combined, i.e. -3r
 multi-part args cannot.
 -3 Enable 3.3V
 -5 Enable 5V
```

maxgerhardt, 2023-06-03:
I see. How about we do this then: Just remove the readme line talking about there being a prebuilt binary and expecting the user to always build themselves. Of course we can CI to test-build `minichlink` as a compilation sanity check.

mengstr, 2023-06-03:
I think that sounds like the best option - at least until github allows for M1/M2 builds. 

But thinking about it again... There might be a some people still using X86 Macs out there. Some people upgrades their machines to the latest and greatest at every opportunity, hopefully I'm not the only one that's a bit more frugal. I used my 2013 MBP as my main machine until 2022.

On the other hand this will frustrate M1/M2 users when they start looking for problems with theirs installation of libusb trying to figure out why their installed version doesn't work. That'd be annoying.

Just get rid of the executables for now I think. ;-)

maxgerhardt, 2023-06-03:
In the folder the binary is already gone, and magically I can just edit the Wiki page (??) so I'll put a little strike-through in there.

Additionally, we can link to the downloadable minichlink binaries built in the CI, just for reference.

maxgerhardt, 2023-06-03:
https://github.com/cnlohr/ch32v003fun/wiki/Installation#macos was corrected.

closed 2023-06-03 by mengstr