cnlohr/ch32fun #151: The serial interface does not work in parallel with the I2C mode

panciunio, 2023-06-04:
Hi Cnlohr,

At first - many thanks for your effort and for all contributors! It's very useful project and I'm very happy using it! I hope it will be available for CH32V103 too :-)

I tried to use an example I2C_OLED with partially success... OLED working as expected but serial output doesn't work - no one message appeared on serial console. To confirm that any connection and configuration is ok, I used other example: UARTDEMO where everything works fine and each message are visible on my console:
[image]

I compared both examples and I can't find any difference between into serial initialization, so I suppose the issue coming somewhere else in rest of code. Could you please advise where I shall to look?

Best Regards,
pancio

maxgerhardt, 2023-06-04:
If you want the debug printf to go out via serial / UART, then you need to activate the macro responsible for it, by default it's commented out

https://github.com/cnlohr/ch32v003fun/blob/b199584bd2d586b91c5a117ff9ec758d092f39d2/examples/i2c_oled/Makefile#L3-L6

Otherwise it tries to push out the debug messages via the flashing interface (SDI), readable with the `minichlink` program.

panciunio, 2023-06-04:
Just minute before I found this #CFLAGS+=-DSTDOUT_UART  definition in Makefile :-) But many thanks for your advice. Could you please explain how to do it with minichlink? Should be data available on /dev/ttyACM0 (my WCH-link) after flashing process?

maxgerhardt, 2023-06-04:
No in the not STDOUT_UART case the data is passed via the debug registers in the debug connection. As the readme says per

https://github.com/cnlohr/ch32v003fun#a-fast-printf-debug-over-the-programming-interface

you need to use `./minichlink -T` to get those prints.

panciunio, 2023-06-04:
Now everything is clear and works fine. There is no possibility to see data from ACM0 device directly on terminal e.g. `minicom` app but  ` ./minichlink -T` is very usable. Is there any example how to catch data from serial port by UART?

maxgerhardt, 2023-06-04:
Hm I didn't see a UART RX example, but if you look at the `_write` implementation and UART init

```
https://github.com/cnlohr/ch32v003fun/blob/b199584bd2d586b91c5a117ff9ec758d092f39d2/ch32v003fun/ch32v003fun.c#L879-L906C2
```

you could come up with an extension of that, that activates the RX mode and implements the `_read` function so that standard-IO functions like `scanf()` or `getchar()` just work, see e.g. https://shawnhymel.com/1873/how-to-use-printf-on-stm32/ for reference.

Let me just test this for myself :)

maxgerhardt, 2023-06-04:
Hmm I got it working in my PlatformIO enviornment but I had to comment out some functions in `ch32v003fun.c` because otherwise it would generate duplicate definitions for standard-C functions.. need to investigate it more. But when I comment those out (as would be mentioned in the compile errors), this program works nicely:

[image]

```c
#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK

#include "ch32v003fun.h"
#include <stdio.h>
#include <unistd.h>
#include <string.h>

int _read(int fd, char *ptr, int len)
{
  if (fd == STDIN_FILENO) {
    // wait for a character to be received
    while(!(USART1->STATR & USART_STATR_RXNE)) {}
    uint8_t rxData = USART1->DATAR;
    *ptr = rxData;
    //always only return 1 byte
    return 1;
  }
  return -1;
}

int main() {
  SystemInit48HSI();
  SetupUART(UART_BRR);
  // we need to additional turn on RX mode
  USART1->CTLR1 |= USART_Mode_Rx;
  Delay_Ms(1000); // startup delay

  while (1) {
    printf("Please enter some text: ");
    // read from stdin FILE* / stream
    char buf[128];
    fgets(buf, sizeof(buf), stdin);
    // will return after hitting newline, but newline chracter is included.
    // beauty massage: overwrite it with NUL character here
    char *pos;
    if ((pos=strchr(buf, '\n')) != NULL)
      *pos = '\0';
    if ((pos=strchr(buf, '\r')) != NULL)
      *pos = '\0';
    printf("You entered: \"%s\"\n", buf);
  }
}
```

panciunio, 2023-06-04:
Of course... I'm rather noob, more concentrated on hardware than software but I see big potential with CH32V003 - so small but so powerfully :-) 
Let me check your example and confirm that working fine... suppose shall be added as additional example.

maxgerhardt, 2023-06-04:
Actually I had a hard time getting it to compile with this makefile / standard toolchain because it doesn't link / have the standard nosys.specs and nano.specs that I'm linking with.

As an intermediate solution I rewrote it to **not** rely on Standard-IO functions, but do this instead:

```c
#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK

#include "ch32v003fun.h"
#include <stdio.h>
#include <unistd.h>
#include <string.h>
#include <stdbool.h>

int _read(int fd, char *ptr, int len)
{
  if (fd == STDIN_FILENO) {
    // wait for a character to be received
    while(!(USART1->STATR & USART_STATR_RXNE)) {}
    uint8_t rxData = USART1->DATAR;
    *ptr = rxData;
    //always only return 1 byte
    return 1;
  }
  return -1;
}
// user input buffer and position in buffer
static char buf[128];
static unsigned pos = 0;
bool get_user_input() {
	char c;
	if(pos < sizeof(buf) && _read(STDIN_FILENO, &c, 1) == 1) {
		// assume terminal program is set to either \r\n or \n as line ending, thus \n is always end of line
		if(c == '\n') {
			return true;
		} 
		// swallow \r (line ending normalization basically)
		if (c == '\r')
			return false;
		buf[pos++] = c;
		// we don't have a full line yet
		return false;
	} else {
		//something bad happened, reset
		pos = 0;
		memset(buf, 0, sizeof(buf));
		return false;
	}
}

int main() {
  SystemInit48HSI();
  SetupUART(UART_BRR);
  // we need to additional turn on RX mode
  USART1->CTLR1 |= USART_Mode_Rx;
  Delay_Ms(1000); // startup delay

  while (1) {
    printf("Please enter some text: ");
    // read from user until newline (blocking)
	while(!get_user_input()) {}
	printf("You entered: \"%s\"\n", buf);
	// reset for next run
	pos = 0;
	memset(buf, 0, sizeof(buf));
  }
}
```
this should compile without problems and still works, and is a significantly smaller firmware as a side-effect.

panciunio, 2023-06-04:
Good job Max,

Working fine:
[image]

So maybe some commit to branch? 

Thanks a lot for help!

Best Regards,

maxgerhardt, 2023-06-04:
(You can turn on local echo mode in miniterm with Ctrl + T followed by Ctrl + E

Great that it works, but let's get this issue back on track. The original question was about printf() not showing up as UART output in the I2C OLED demo, but we resolved that with the unactivated `STDOUT_UART` macro.

Adding a proper UART RX is another point, I've opened #152 for that.

closed 2023-06-04 by panciunio