cnlohr/ch32fun #152: Add UART RX demo

maxgerhardt, 2023-06-04:
examples/uartdemo is currently TX only, the `ch32v003fun.c` code also initializes the UART in TX only mode.

https://github.com/cnlohr/ch32v003fun/blob/b199584bd2d586b91c5a117ff9ec758d092f39d2/ch32v003fun/ch32v003fun.c#L889-L891

It would be nice to add support for UART RX and interfacing it with standard I/O functions, so probably `getc()` / `getchar()` /  `fgets()`.

See some example code at https://github.com/cnlohr/ch32v003fun/issues/151#issuecomment-1575556714, this uses its own user-made function instead of stdio functions.

cnlohr, 2023-06-05:
I would be up for adding another #define to enable more functionality on the UART. That could be defined in the Makefile.  I don't have an immediate need.  I know @dwillmore has played with UART in some.

karstengit, 2023-06-20:
> It would be nice to add support for UART RX and interfacing it with standard I/O functions, so probably `getc()` / `getchar()` / `fgets()`.

It would be really fine to have an example how to read data from the UART.

karstengit, 2023-06-20:
Specially it would be interesting to have an example how to configure an interrupt routine filling an string buffer.

[image]

This is the example from the MRS:

```
/*********************************************************************
 * @fn      USARTx_CFG
 *
 * @brief   Initializes the USART2 & USART3 peripheral.
 *
 * @return  none
 */
void USARTx_CFG(void)
{
    GPIO_InitTypeDef  GPIO_InitStructure = {0};
    USART_InitTypeDef USART_InitStructure = {0};
    NVIC_InitTypeDef  NVIC_InitStructure = {0};

    RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD | RCC_APB2Periph_USART1, ENABLE);

    /* USART1 TX-->D.5   RX-->D.6 */
    GPIO_InitStructure.GPIO_Pin = GPIO_Pin_5;
    GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;
    GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF_PP;
    GPIO_Init(GPIOD, &GPIO_InitStructure);
    GPIO_InitStructure.GPIO_Pin = GPIO_Pin_6;
    GPIO_InitStructure.GPIO_Mode = GPIO_Mode_IN_FLOATING;
    GPIO_Init(GPIOD, &GPIO_InitStructure);

    USART_InitStructure.USART_BaudRate = 115200;
    USART_InitStructure.USART_WordLength = USART_WordLength_8b;
    USART_InitStructure.USART_StopBits = USART_StopBits_1;
    USART_InitStructure.USART_Parity = USART_Parity_No;
    USART_InitStructure.USART_HardwareFlowControl = USART_HardwareFlowControl_None;
    USART_InitStructure.USART_Mode = USART_Mode_Tx | USART_Mode_Rx;

    USART_Init(USART1, &USART_InitStructure);
    USART_ITConfig(USART1, USART_IT_RXNE, ENABLE);

    NVIC_InitStructure.NVIC_IRQChannel = USART1_IRQn;
    NVIC_InitStructure.NVIC_IRQChannelPreemptionPriority = 1;
    NVIC_InitStructure.NVIC_IRQChannelSubPriority = 1;
    NVIC_InitStructure.NVIC_IRQChannelCmd = ENABLE;
    NVIC_Init(&NVIC_InitStructure);;

    USART_Cmd(USART1, ENABLE);
}

void USART1_IRQHandler(void) __attribute__((interrupt("WCH-Interrupt-fast")));

/*********************************************************************
 * @fn      USART1_IRQHandler
 *
 * @brief   This function handles USART3 global interrupt request.
 *
 * @return  none
 */
void USART1_IRQHandler(void)
{
    if(USART_GetITStatus(USART1, USART_IT_RXNE) != RESET)
    {
        RxBuffer1[RxCnt1++] = USART_ReceiveData(USART1);

        if(RxCnt1 == TxSize1)
        {
            USART_ITConfig(USART1, USART_IT_RXNE, DISABLE);
            Rxfinish1 = 1;
        }
    }
}

```

cnlohr, 2023-06-21:
Would you be up for writing one? / Working on this?

karstengit, 2023-06-21:
> Would you be up for writing one? / Working on this?

Yes - when there is a working example it will be posted here.
But this will take much time, because the bad programmer is very slow and need the right mood to get into it ...
(Maybe your question will help to motivate :-)

Is it possible to upload in this github repository?
(But this is an additional challenge :-)

karstengit, 2023-06-21:
Some open thinking:

How it is possible to resolve the existing definitions in ch32v003fun.h ?
Searching for a tool for it only is analyzing other dependencies like the header files.

```
/* Universal Synchronous Asynchronous Receiver Transmitter */
typedef struct
{
    __IO uint16_t STATR;
    uint16_t      RESERVED0;
    __IO uint16_t DATAR;
    uint16_t      RESERVED1;
    __IO uint16_t BRR;
    uint16_t      RESERVED2;
    __IO uint16_t CTLR1;
    uint16_t      RESERVED3;
    __IO uint16_t CTLR2;
    uint16_t      RESERVED4;
    __IO uint16_t CTLR3;
    uint16_t      RESERVED5;
    __IO uint16_t GPR;
    uint16_t      RESERVED6;
} USART_TypeDef;

```

The different interrupts in CTLR1 seems not to be defined.
Regarding the above list of interrupts a search for "TXNE" does not give any result in any file.

It can only be found:

    USART1_IRQn = 32,        /* USART1 global Interrupt                              */

In the MRS example this function is used to determine the interrupt:

```
/*********************************************************************
 * @fn      USART_GetITStatus
 *
 * @brief   Checks whether the specified USART interrupt has occurred or not.
 *
 * @param   USARTx - where x can be 1 to select the USART peripheral.
 *          USART_IT - specifies the USART interrupt source to check.
 *            USART_IT_LBD - LIN Break detection interrupt.
 *            USART_IT_TXE - Tansmit Data Register empty interrupt.
 *            USART_IT_TC - Transmission complete interrupt.
 *            USART_IT_RXNE - Receive Data register not empty interrupt.
 *            USART_IT_IDLE - Idle line detection interrupt.
 *            USART_IT_ORE_RX - OverRun Error interrupt if the RXNEIE bit is set.
 *            USART_IT_ORE_ER - OverRun Error interrupt if the EIE bit is set.
 *            USART_IT_NE - Noise Error interrupt.
 *            USART_IT_FE - Framing Error interrupt.
 *            USART_IT_PE - Parity Error interrupt.
 *
 * @return  bitstatus: SET or RESET.
 */
ITStatus USART_GetITStatus(USART_TypeDef *USARTx, uint16_t USART_IT)
{
    uint32_t bitpos = 0x00, itmask = 0x00, usartreg = 0x00;
    ITStatus bitstatus = RESET;

    usartreg = (((uint8_t)USART_IT) >> 0x05);
    itmask = USART_IT & IT_Mask;
    itmask = (uint32_t)0x01 << itmask;

    if(usartreg == 0x01)
    {
        itmask &= USARTx->CTLR1;
    }
    else if(usartreg == 0x02)
    {
        itmask &= USARTx->CTLR2;
    }
    else
    {
        itmask &= USARTx->CTLR3;
    }

    bitpos = USART_IT >> 0x08;
    bitpos = (uint32_t)0x01 << bitpos;
    bitpos &= USARTx->STATR;

    if((itmask != (uint16_t)RESET) && (bitpos != (uint16_t)RESET))
    {
        bitstatus = SET;
    }
    else
    {
        bitstatus = RESET;
    }

    return bitstatus;
}

```

This can be found in ch32v003fun.h too:

```
/* USART_Interrupt_definition */
#define USART_IT_PE                          ((uint16_t)0x0028)
#define USART_IT_TXE                         ((uint16_t)0x0727)
#define USART_IT_TC                          ((uint16_t)0x0626)
#define USART_IT_RXNE                        ((uint16_t)0x0525)
#define USART_IT_ORE_RX                      ((uint16_t)0x0325)
#define USART_IT_IDLE                        ((uint16_t)0x0424)
#define USART_IT_LBD                         ((uint16_t)0x0846)
#define USART_IT_CTS                         ((uint16_t)0x096A)
#define USART_IT_ERR                         ((uint16_t)0x0060)
#define USART_IT_ORE_ER                      ((uint16_t)0x0360)
#define USART_IT_NE                          ((uint16_t)0x0260)
#define USART_IT_FE                          ((uint16_t)0x0160)

```

O.K. So the use of this functions seems to be the easiest solution at this time.

karstengit, 2023-06-21:
> Would you be up for writing one? / Working on this?

Yes - when you are patient enough ... :-)

First I should solve the problem with the LCD display to get more familiar with this MCU.

dwillmore, 2023-06-25:
> Somebody has added an example *hsitrim* that shows how to read an char
> from the UART:
>
> if(USART1->STATR & USART_STATR_RXNE){
> 	uint8_t key = (uint8_t)USART1->DATAR;
> 	if(key == '.') { ... }

That's me!  I think I may have written the first UART read code, but it
wasn't the point of the example, so I didn't dwell on it much.  We could
use a good example (and maybe an expansion library) to make it more useful
for others.  Note we also have debug port read, IIRC.  Which you may find
useful.

karstengit, 2023-06-25:
Hi [dwillmore](https://github.com/dwillmore), I think you will write such a library in a fraction of time as myself.
My next step is to understand how to use interrupts and the timers.

> Note we also have debug port read, IIRC.  Which you may find useful.

When I know what it is I maybe will find it useful.
You will laugh, but as "programmer of simplicity" I never used the gdb debug interface.
I program step by step simple subroutines, that can easily be tested and some debug prints over the UART where always enough.

hexeguitar, 2023-07-23:
I recently wrote this single header UART library for the '003:
https://gist.github.com/hexeguitar/a97b3ec2764c845c4377487368ddad33
Currently implemented:
- configurable TX, RX or TX+RX
- alternative pin mapping
- disables itself if FUNCONF_USE_UARTPRINTF is used, leaving the USART1 for the debug channel
- uses the FUNCONF_UART_PRINTF_BAUD to set the baudrate
- 8N1, no parity, no hw flow control
- buffered TX and RX using interrupts
- can "printf" the same way as debug channel

I can submit a PR if you think it could be added to the extra_libs.

mengstr, 2023-07-24:
Maybe I'm too oldskool to understand, but why do people put code in header files nowadays? Originally .h files was for function declarations, defines and macros. Not code.

One of the tasks of the linker is to join multiple object files together. There's no need to #include all sources into one single file when compiling.

But I guess there's a valid reson for this trend, it has just passed me by. ;-)

karstengit, 2023-07-24:
I promise you are not oldskool enough. :-)

For this really small controllers I always include the sources direct into the main file, because compilation only needs a part of a second.
It is understandable to use the better structure of sources and header files for bigger projects or for the ESP MCU's, where a ton of libraries are always compiled and linked together and you can really save time when you only alter one file.

By the way - is there any difference on the file ending for the compiler?
I think the compiler handle everything the way it is directed with the given options and not by the file ending.
So Arduino has choosen to name there files **ino** to make a difference as maker. ;-)

But I agree that I would not expect any code to be compiled in an **.h** header file.
In this case I would rename it and would be still happy to get working code.

cnlohr, 2023-07-31:
> 

It may make sense to be part of examples, if it can be paired down.  Specifically for serial.

In general you should be using this facility:

```c
// Call this to busy-wait the polling of input.
void poll_input();

// Receiving bytes from host.  Override if you wish.
void handle_debug_input( int numbytes, uint8_t * data );
```

Other than that, it is time to close this issue.

closed 2023-07-31 by cnlohr