cnlohr/ch32fun #174: Proposal: Add GPIO operational on whole port

panciunio, 2023-06-15:
HI,

There is very good reason to use CH32V003 as successor for ATMEGA family, so it will be very helpfully to have method to read, write and modify on whole port e.g.:
`GPIOD_OUTDR = 0b10111111; all bits set as PULL-UP input
GPIOA_BSHR = 0b10111111; reset all bits in PA
GPIOC_BSHS = 0b10110000; set four MSB bits in PC
`
I know that it's possible using defined bu user structure but above is more easier...

cnlohr, 2023-06-15:
I don't understand your proposal.  Are you asking for #defines for these symbols?  Are they not just `#define GPIOD_OUTDR GPIOD->OUTDR`?

panciunio, 2023-06-15:
No... I'm thinking about little easier method for set/reset bits in whole port at the same time using values 0...255. 
Now we have construction like.:
```
// GPIO D0 Push-Pull
	GPIO_pinMode(GPIO_port_D, 0, GPIO_pinMode_O_pushPull, GPIO_Speed_10MHz);
```
for one bit only... but I want to set whole port at once.

Other example:
```
		GPIO_digitalWrite(GPIO_port_D, 0, high);
		GPIO_digitalWrite(GPIO_port_D, 4, high);
		GPIO_digitalWrite(GPIO_port_C, 0, high);
```

maybe this method are able to set three bits in the same time... (depend on compiler optimization?),  but I want to have construction like this : 
`GPIO_digitalWrite(GPIO_port_D, 0b00010001); //set bit 4 and 0 in port D`
or
`GPIO_digitalWrite(GPIO_port_D, 0x11); //set bit 4 and 0 in port D`

Why? This method will provide information about all bits in the same time - it's very usable if you are using port as register for external bus...

panciunio, 2023-06-18:
I found two solution for this inconvenience... 

`#define GPIO_digitalRead8(GPIO_port_n, pin)	 	((GPIO_port_n_to_GPIOx(GPIO_port_n)->INDR >> pin) & 0b11111111)`
then use:
`data = GPIO_digitalRead8(GPIO_port_C, 0);`

or use:
`GPIO_INDR_t data2 = DYN_GPIO_READ(GPIOC, INDR);`

seems, that result is the same... 8 bita are reading at once:

```
         GPIO_INDR_t data2 = DYN_GPIO_READ(GPIOC, INDR);
 5c0:	400117b7          	lui	a5,0x40011
 5c4:	478c                	lw	a1,8(a5)

```

panciunio, 2023-06-18:
Working code:

```
// Reading data directly from PORTC and sending on SERIAL in hex format.
// D3: (D5_is_pressed) chip select 
// D4: (RW_low) - Read/Write (active low)

// Could be defined here, or in the processor defines.
#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK

#include "ch32v003fun.h"
#include <stdio.h>

#include "ch32v003_GPIO_branchless.h"

int main()
{
	SystemInit48HSI();
//	SetupUART( UART_BRR );   // 115200 baud
	SetupUART( UART_BRR/8 ); // 921600 baud

	// Enable GPIOD and GPIOC    
    GPIO_portEnable(GPIO_port_C);
    GPIO_portEnable(GPIO_port_D);

	while(1)   	
    {
        
        uint8_t D5_is_pressed = !GPIO_digitalRead(GPIO_port_D, 3);
        if (D5_is_pressed)
        {
            Delay_Us(0.1);
            uint8_t RW_low = !GPIO_digitalRead(GPIO_port_D, 4);
            if (RW_low)
            {
                GPIO_INDR_t data2 = DYN_GPIO_READ(GPIOC, INDR);
                printf("%02x", data2);
            }
            
        }     
	}
}

```
How to send data directly to SERIAL without extremally slowly _printf()_?

recallmenot, 2023-06-21:
https://github.com/cnlohr/ch32v003fun/pull/188

cnlohr, 2023-06-22:
Going back to some earlier portions - I still don't understand why you would want `x = GPIO_digitalRead(GPIO_port_D);` instead of `x = PORTD->INDR`?

It just seems clearer that one is a register read where the other "might be a function maybe?"

panciunio, 2023-06-22:
Hi,

Actually.. I'm using INDR/OUTDR in my project now. Earlier I didn't realize that it's possible because not find proper example with direct access to port based on 8-bit data. So construction like working perfect! : 

```
uint8_t data2 = GPIOC->INDR;   //catch data from PORTC
GPIOC->OUTDR = 0xff; // sent 255 to PORTC
```
It was a  lack of my knowledge, now I know how to use registers directly. Topic closed. 

According to serial write: https://github.com/cnlohr/ch32v003fun/issues/174#issuecomment-1600373300

`putchar(data); `

it is best method for me (next to pointed by you) - many thanks!

Thanks a lot!,
pancio

cnlohr, 2023-08-02:
I am going to close this issue now.  I believe it is all cleared up.

closed 2023-08-02 by cnlohr