cnlohr/ch32fun #181: NRST mention in README

mengstr, 2023-06-18:
The readme says `Optional (not needed, can be configured as output if fuse set): NRST is on PD7`. 
To avoid unnecessary confusion and datasheet reading It might be good to mention that PD7 by default is a GPIO.

[image]

cnlohr, 2023-06-19:
Do you have a proposal on how to convey it clearly?

mengstr, 2023-06-19:
LOL no...  As a non-native English speaker that recently, even for my own mother tongue, got low scores on "clarity of communication" in the yearly evaluation at my workplace I might not be the best person to ask :-)

Maybe just invert the context of the sentence? 
`Optional: NRST on PD7 (not needed, but be configured if the RST_MODE fuses in flash are modified) `

cnlohr, 2023-06-19:
I rejiggerd it some.  I think it's good - closing issue now.

closed 2023-06-19 by cnlohr