cnlohr/ch32fun #182: Purpose of this project?

mengstr, 2023-06-18:
I thought that this project was mostly about moving away from a (the official) HAL and go more towards a direct to the metal approach, so the discussions about GPIO abstractions and functions for easier access to them makes me a bit confused since it feels very Arduino'esqe.

Is this project going slowly towards an Arduino-styled HAL or will it stay metal with just small demo/example codes for doing common tasks?

cnlohr, 2023-06-18:
As many examples as possible should use the register functionality

Only for people who are unwilling to use the register functionality should they use the arduino stuff.

The arduino stuff is made available to aid in porting things to this processor as well.

prosper00, 2023-06-18:
Well, there is this, too

https://github.com/openwch/arduino_core_ch32

I haven't tried it, as it's currently Windows-only

cnlohr, 2023-06-21:
This is still going back mostly to the GPIO thing I think, for discussion sake from the discord:

Going back to this discussion, which I really want to have.  I think my biggest issue is this whole `_branchless` stack is very heavy-handed.  And has a lot of gotchas depending on the device hiding within it.  My goal would be to minimize its complexity. 

I think I need to spend some time and come up with a proposal to try to articulate what I am thinking, because right now, all I'm doing is a bunch of hand waving and maybe there's some serious problem deep within my thinking on it that can only be discovered by doing.  I was thinking a stack like this, for the STM32... 

https://github.com/cnlohr/stm32f042_fun/blob/master/stm32f042cXt6_demo/firmware/systems.h#L33-L99

Or, maybe exactly that, just for the 042.  What it does is it uses a `uint8_t` to define a port and pin. The top 4 bits determin the port ID (A, B, C, etc.) and the bottom 4 define the pin number inside that port.  (Note: Probably will not want to include `GetGPIOFromString` and I would want to macroify the configuration.

It lets you use things like:
```c
GPIOOn(0x10);  //Port B0 tuurn on
ConfigureGPIO( GPIO_FROM_NUMS( 0, 7 ), 0b11 ); // GPIO_FROM_NUMS( port, pin )
int r = GPIOOf(GPIO_FROM_NUMS( PORT_C, 0 ))->IDR; // Read all inputs from Port C
```

THOUGH The fundamental issue here is that you lose a lot by doing a bunch of `|=`'s instead of just setting up a port, like this:
```c
    GPIOC->CFGLR = (GPIO_Speed_50MHz | GPIO_CNF_OUT_PP)<<(4*0) |
                   (GPIO_Speed_50MHz | GPIO_CNF_OUT_PP_AF)<<(4*3) | // PC3 = T1C3
                   (GPIO_Speed_50MHz | GPIO_CNF_OUT_PP_AF)<<(4*4) | // PC4 = T1C4
                   (GPIO_Speed_50MHz | GPIO_CNF_OUT_PP)<<(4*2);
```

Like I don't want to teach people to use a library, when the chip's power and simplicity is unlocked if I can get them to bite the bullet and just use the registers the way they were intended to be used.

Re: wildwest, I was starting to do some experiments with the SSD1306 in the wildwest repo, (in my fork of it) but, it ended up feeling really weird, because I wasn't re-using any of the existing tools inside wildwest because they didn't quite map to my needs, i.e. I brought in my `static_i2c.h` but then I feel weird/dumb for not re-using or improving the existing libraries in wildwest.  I find it very hard for me to figure out how to meaningfully contribute and work in environments with lots of libraries.

eeucalyptus, 2024-09-13:
Since there is a macro based `fun*` gpio interface in the lib, I consider this issue as obsolete

closed 2024-09-13 by eeucalyptus