cnlohr/ch32fun #191: Add deep sleep example

cnlohr, 2023-06-21:


recallmenot, 2023-06-22:
Isn't standby the lowest power mode:
https://github.com/cnlohr/ch32v003fun/tree/master/examples/standby_autowake

cnlohr, 2023-06-23:
Actually yes.  It is.  Could definitely use more comments explaining it.  I am not sure how to apply very much of it.  I can work from there.  Thank you for pointing that out.

cnlohr, 2023-06-23:
I guess I'll close for now.

recallmenot, 2023-06-24:
https://github.com/recallmenot/ch32v003fun_wildwest/tree/main/proj/standby_autowake_and_btn
https://github.com/recallmenot/ch32v003fun_wildwest/blob/main/lib/ch32v003_standby.h
might these be helpful?

cnlohr, 2023-06-29:
Reopening until it's solved - these seem to use 100-200uA?

eeucalyptus, 2024-09-13:
Telling by the git history, the problem was a missing power cycle after programming (see https://github.com/cnlohr/ch32v003fun/commit/8b05efb3d8254084cc1eabf40d54b32319cf1219)

closed 2024-09-13 by eeucalyptus