cnlohr/ch32fun #192: No GPIO access works, but complexer functions

karstengit, 2023-06-23:
Sorry for the next chaotic bug report.

I have a written a simple demo that loads the shift register with latch over the SPI.
The crazy thing is that the SPI works, but with double speed, so SCK has 25 MHz instead of 10 MHz.
MOSI is showing a correspondend data output.
But the shifted data could not be latched, because the GPIO pin access does not work!

So I did go back to your GPIO example and there is no Port activity.
To be sure that it is not my breakout board I use now the original WCH board.
Again no port activity.

Here is the compiler output:

```
$ make
riscv64-unknown-elf-gcc -o GPIO.elf ../../ch32v003fun/ch32v003fun.c GPIO.c wiring.c -DTINYVECTOR -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
riscv64-unknown-elf-size GPIO.elf
   text    data     bss     dec     hex filename
    672       0       4     676     2a4 GPIO.elf
riscv64-unknown-elf-objdump -S GPIO.elf > GPIO.lst
riscv64-unknown-elf-objdump -t GPIO.elf > GPIO.map
riscv64-unknown-elf-objcopy -O binary GPIO.elf GPIO.bin
riscv64-unknown-elf-objcopy -O ihex GPIO.elf GPIO.hex
make -C ../../minichlink all
make[1]: Verzeichnis „/win/WCH/ch32v003fun/minichlink“ wird betreten
make[1]: Für das Ziel „all“ ist nichts zu tun.
make[1]: Verzeichnis „/win/WCH/ch32v003fun/minichlink“ wird verlassen
../../minichlink/minichlink -w GPIO.bin flash -b
Found WCH Link
WCH Programmer is LinkE version 2.8
Chip Type: 003
Part Type (A): 0x0010 (This is the capacity code, in KB)
Part UUID    : 39-ca-ab-cd-a1-d8-bc-41
PFlags       : ff-ff-ff-ff
Part Type (B): 00-30-05-00
Interface Setup
Image written.
```

The gcc is

```
$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/8.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' LDFLAGS='-Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 8.3.0 () 
```

Installed as package gcc-riscv64-unknown-elf 8.3.0.2019.08+dfsg-1 in Debian GNU/Linux 11 (bullseye).

The blink program compiled in the Moun River Studio works.
So what is going on?

karstengit, 2023-06-23:
If needed here is a zip of the complete GPIO directory.
[GPIO.zip](https://github.com/cnlohr/ch32v003fun/files/11851532/GPIO.zip)

alexhorner, 2023-06-23:
> 

Hey karstengit, are you using the SPI's controlled CS access, or toggling CS manually using the registers yourself? I have found that the built in SPI control CS didn't work (but not something I have tried to diagnose) and opted to manually control the CS GPIO myself.

karstengit, 2023-06-23:
Thanks for the quick response.
In my example I tried manually to toggle the CS and after this did not work in any case I tried the GPIO example.

Here is my example:

```
/*
 * Example for using SPI with a 74HC595 as port expander
 * 
 * Measured:
 * SCK 25 MHz
 * 
 * 
 * 06-23-2023 by karsten
 */

// Could be defined here, or in the processor defines.
#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK

// Pin for chip select
#define LATCH_CS_PORT GPIOC
#define LATCH_CS_PIN 0
#define LATCH_CS_HIGH() LATCH_CS_PORT->BSHR = (1 << (LATCH_CS_PIN))
#define LATCH_CS_LOW() LATCH_CS_PORT->BSHR = (1 << (16 + LATCH_CS_PIN))

#include "ch32v003fun.h"
#include <stdio.h>

int count = 0;

/*
 * init SPI and GPIO
 */
void spi_init(void)
{
	// Enable CS pin
	LATCH_CS_PORT->CFGLR &= ~(0xf << (4 * LATCH_CS_PIN));
	LATCH_CS_PORT->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP) << (4 * LATCH_CS_PIN);
	LATCH_CS_LOW();

	// Enable GPIOC and SPI
	RCC->APB2PCENR |= RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1;

	// PC5 is SCK, 10MHz Output, alt func, p-p (pin 11 74HC595)
	GPIOC->CFGLR &= ~(0xf << (4 * 5));
	GPIOC->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP_AF) << (4 * 5);
	
	// PC6 is MOSI, 10MHz Output, alt func, p-p  (pin 14 74HC595, pin 10 = vcc, pin 13 = gnd)
	GPIOC->CFGLR &= ~(0xf << (4 * 6));
	GPIOC->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP_AF) << (4 * 6);
	
	// Configure SPI 
	SPI1->CTLR1 = 
		SPI_NSS_Soft | SPI_CPHA_1Edge | SPI_CPOL_Low | SPI_DataSize_8b |
		SPI_Mode_Master | SPI_Direction_1Line_Tx |
		SPI_BaudRatePrescaler_2;

	// enable SPI port
	SPI1->CTLR1 |= CTLR1_SPE_Set;
}

int main()
{
	// 48MHz internal clock
	SystemInit48HSI();

	// start serial @ default 115200bps
#ifdef STDOUT_UART
	SetupUART( UART_BRR );
	Delay_Ms( 100 );
#else
	SetupDebugPrintf();
#endif
	printf("\nexample for load of serial latch\n\n");

	// init spi and oled
	Delay_Ms( 100 );	// give OLED some more time
	printf("initializing spi oled...");

	spi_init();
	printf("spi init done.\n");
		
	printf("Looping max. speed test output...");
	while(1)
	{
		SPI1->DATAR = 0b10101010;
		while(SPI1->STATR & SPI_STATR_BSY);	// wait for not busy
		LATCH_CS_HIGH();
		LATCH_CS_LOW();

		SPI1->DATAR = 0b01010101;
		while(SPI1->STATR & SPI_STATR_BSY);	// wait for not busy
		LATCH_CS_HIGH();
		LATCH_CS_LOW();

// 		printf("count = %d\n\r", count++);
// 		Delay_Ms(2000);
	}
}
```

alexhorner, 2023-06-23:
> Thanks for the quick response. In my example I tried manually to toggle the CS and after this did not work in any case I tried the GPIO example.
>
> Here is my example:
>
> ```
> /*
>  * Example for using SPI [...]

I definitely have SPI working, if you take a look at PR #153 you'll find my library using it which you can use as reference, and you'll also see I made some changes to the SPI library itself. I am going to change my PR so that the SPI changes are separate now to make sure they get accepted, as my driver is still under discussion, stand by!

karstengit, 2023-06-23:
> I definitely have SPI working, if you take a look at PR #153 you'll find my library using it which you can use as reference, and you'll also see I made some changes to the SPI library itself. I am [...]

Please give me a tip how to clone your library from github (I am somehow not compatible with it.)
A complete clone seems not to include it.

alexhorner, 2023-06-23:
> There is a new gpio example **direct_gpio** [direct_gpio.zip](https://github.com/cnlohr/ch32v003fun/files/11852476/direct_gpio.zip)
> 
> **This works and the GPIO output is working!**

Ah okay no worries, it was gonna take me a little bit to get that in as an extra PR

karstengit, 2023-06-23:
Problem found!

It is the line:
```
	// Enable GPIOC and SPI
	RCC->APB2PCENR |= RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1;
```

There where missing brackets around the or command
```
	// Enable GPIOC and SPI
	RCC->APB2PCENR |= (RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1);
```

alexhorner, 2023-06-23:
> Problem found in my example!
> 
> It is the line:
> 
> ```
> 	// Enable GPIOC and SPI
> 	RCC->APB2PCENR |= RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1;
> ```
> 
> There where missing brackets around the or
> 
> ```
> 	// Enable GPIOC and SPI
> 	RCC->APB2PCENR |= (RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1);
> ```

Interesting... Can you try without them to confirm? I don't think they should make any difference

karstengit, 2023-06-23:
> Interesting... Can you try without them to confirm? I don't think they should make any difference

Yes - this was definitely the problem.

And there are other curiosities.
Compiling the code with
```
		SPI1->DATAR = 0b10101010;
		while(SPI1->STATR & SPI_STATR_BSY);	// wait for not busy
		GPIOC->BSHR = 1 << (0);			// SET GPIO C0
		GPIOC->BSHR = (1 << (16 + 0));	// RESET GPIO C0
```
has an slower output (1256 kHz) as compiling with

```
#define LATCH_CS_HIGH() LATCH_CS_PORT->BSHR = (1 << (LATCH_CS_PIN))
#define LATCH_CS_LOW() LATCH_CS_PORT->BSHR = (1 << (16 + LATCH_CS_PIN))
		SPI1->DATAR = 0b10101010;
		while(SPI1->STATR & SPI_STATR_BSY);	// wait for not busy
		LATCH_CS_HIGH();
		LATCH_CS_LOW();
```

that is changing the port expander with 1444 kHz.

The SPI clock is still 25 MHz. Why?

Setting this bug to completed, because the main problem is identified.

closed 2023-06-23 by karstengit

recallmenot, 2023-06-24:
@karstengit 
ok so I did a bit of digging:
I traveled back to before the branchless GPIO library `git checkout 4f48d78`
and discovered your wiring.c had a wrong line 46
`GPIOx = GPIOA+(pin>>3);`
and it's supposed to be
`GPIOx = (GPIO_TypeDef *)(((uint8_t*)(GPIOA))+(pin>>3)*0x0400);`
With that, it works, which makes sense since jumps between port registers are rather big (lots of empty space in-between).
The version you have is from here
https://github.com/cnlohr/ch32v003fun/blame/e1a45dbccd24396588b5966012f85ce8b19b74a5/examples/GPIO/wiring.c
And it really was broken (shit happens).
this apparently got solved here:
https://github.com/cnlohr/ch32v003fun/pull/129
and here:
https://github.com/cnlohr/ch32v003fun/pull/131

But all of those lie so far in the past, I'd suggest you sync your github fork master and do a git pull.
We're already 1.5 versions of GPIO library ahead, the latest branchless library is vastly superior to the old branching one.

Also, for `RCC->APB2PCENR |= (RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1);` helping on your example, what version of gcc are you using? You can check with `gcc -v`.  
For me, gcc 12 was a bit buggy: a program with a switch case statement would 8x in size when I added the 5th case, pushing it over the 16kb boundary.

karstengit, 2023-06-24:
Thank you for your investigation.

I am able to follow your findings only without the github stuff.

> @karstengit ok so I did a bit of digging 
> But all of those lie so far in the past, I'd suggest you sync your github fork master and do a git pull. We're already 1.5 versions of GPIO library ahead, the latest branchless library is vastly superior to the old branching one.

Can you please give concrete git commands how to check out the relevant version.

> Also, for `RCC->APB2PCENR |= (RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1);` helping on your example, what version of gcc are you using?

You find it in the first post:

```
$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/8.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' LDFLAGS='-Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3RxXtW/gcc-riscv64-unknown-elf-8.3.0.2019.08+dfsg=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 8.3.0 () 
```

The stable distribution always is some versions behind ...

> You can check with `gcc -v`. For me, gcc 12 was a bit buggy: a program with a switch case statement would 8x in size when I added the 5th case, pushing it over the 16kb boundary.

I am a friend of stability and simplicity and so it is to much stress to try to be on the pulse of time.
That's the reason why i am not familiar with github and other build system. I always have only a folder "versions" in my projects and this was enough up to now for my simple projects. :-)
That will not mean that I don't want to learn.
So how I can add an example to this nice project here?

recallmenot, 2023-06-24:
risc-v, especially with EC extension as used here really is "bleeding edge" and a newer compiler (gcc) will do a better job.
as is this repo, there is no "stable" version because it's still all work-in-progress.
up to a couple of months ago I also was using zip files for versioning projects but man, I can tell you, learning `git` really pays off, especially when working with repositories like this one.
[a good starting point](https://www.youtube.com/watch?v=USjZcfj8yxE&pp=ygUDZ2l0)
git is a tool, github is the website you use the tool with, but git is not limited to online-use / tied strictly to github.
Likely you haven't created your own github fork of this repository yet, which is a requirement to contribute. When you think you have something useful and good enough to be published here, you may look up how to:
 * create a fork
 * commit on your fork
 * do a pull request
I'm mentioning this so you'll later know where to look.
For now, open a terminal in the main directory of ch32v003fun on your pc and run `git pull`.
If you haven't made any modifications in the directory, this will 100% get you the latest version of ch32v003fun.
If you did modify, it's likely going to complain. Go to the parent dir of `ch32v003fun`, rename `ch32v003fun` to `ch32v003fun_old`, open a terminal in the parent dir and run `git clone https://github.com/cnlohr/ch32v003fun.git` to download a fresh copy. That's the link you get when you click on the top-right green button of a repository.
Like you see here in github, there will only be one GPIO example and it references `ch32v003_GPIO_branchless.h` in extralibs.

karstengit, 2023-06-24:
Thank you for your description.

> up to a couple of months ago I also was using zip files for versioning projects but man, I can tell you, learning `git` really pays off, especially when working with repositories like this one.

Yes - that's all right.
But I always get very quick to my limits and have to find out, that other people are much fitter and quicker.
Just spend much time to try to adapt an simple LCD library to use with this MCU without success.
Now I need some time to recover ... :-)

recallmenot, 2023-06-25:
@karstengit
I wish you strength to continue learning as real learning always happens right on the edge of your abilities, don't get discouraged by other people being "better", it only means they've already spent more time in the uncomfortable zone of "what am I doing, this is hard".
Spending time in your "I suck"-zone is the most valuable thing you can do because that's where the greatest learning happens!

karstengit, 2023-06-25:
Will the force of programming always be with you!