cnlohr/ch32fun #193: How to use <stdlib.h> ?

karstengit, 2023-06-24:
I tried to use the <stdlib.h> for simple functions like itoa, utoa, but this does not work.

Let us take this shortened example:
```
// Could be defined here, or in the processor defines.
#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK

#include "ch32v003fun.h"
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

uint32_t count = 42;
char lpuffer[17] ;		//< string buffer for LCD-line
ultoa(count, lpuffer, 10);

int main() {
	while(1) {
}
```

This leads to an undefined reference to `ultoa'

```
$ make build
riscv64-unknown-elf-gcc -o spi_lcd.elf ../../ch32v003fun/ch32v003fun.c spi_lcd.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
spi_lcd.c: In function 'main':
spi_lcd.c:34:3: warning: implicit declaration of function 'ultoa'; did you mean 'utoa'? [-Wimplicit-function-declaration]
   ultoa(count, lpuffer, 10);
   ^~~~~
   utoa
/usr/lib/riscv64-unknown-elf/bin/ld: /tmp/ccN4ShTe.ltrans0.ltrans.o: in function `.L0 ':
/win/WCH/K/examples/spi_lcd/lcd.c:110: undefined reference to `ultoa'
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:24: spi_lcd.elf] Fehler 1
```

or with utoa

```
$ make build
riscv64-unknown-elf-gcc -o spi_lcd.elf ../../ch32v003fun/ch32v003fun.c spi_lcd.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
/usr/lib/riscv64-unknown-elf/bin/ld: /tmp/ccUvYmUs.ltrans0.ltrans.o: in function `.L0 ':
/win/WCH/K/examples/spi_lcd/lcd.c:110: undefined reference to `utoa'
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:24: spi_lcd.elf] Fehler 1
```

What must be done?

karstengit, 2023-06-25:
Alternative I tried to use sprintf to format an values into a string.
But only printf is available, sprintf cannot be found!

```
$ make
riscv64-unknown-elf-gcc -o spi_lcd.elf ../../ch32v003fun/ch32v003fun.c spi_lcd.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
/usr/lib/riscv64-unknown-elf/bin/ld: /tmp/cccOzuwj.ltrans0.ltrans.o: in function `.L0 ':
/win/WCH/K/examples/spi_lcd/lcd.c:110: undefined reference to `sprintf'
/usr/lib/riscv64-unknown-elf/bin/ld: /win/WCH/K/examples/spi_lcd/lcd.c:167: undefined reference to `sprintf'
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:24: spi_lcd.elf] Fehler 1
```

**undefined reference to `sprintf'**

Every conversion of a number into a string is not allowed?
Only printf to an UART?

karstengit, 2023-06-25:
If someone else has the same problem and this bug will not be solved, you can copy as workaround the needed functions into your code.
https://github.com/vancegroup-mirrors/avr-libc/tree/master/avr-libc/tests/simulate/stdlib

For example:
```
static char *ultoa_recursive (unsigned long val, char *s, unsigned radix)
{
    int c;

    if (val >= radix)
	s = ultoa_recursive (val / radix, s, radix);
    c = val % radix;
    c += (c < 10 ? '0' : 'a' - 10);
    *s++ = c;
    return s;
}

static char *ultoa (unsigned long val, char *s, int radix)
{
    if (radix < 2 || radix > 36)
	s[0] = 0;
    else
	*ultoa_recursive (val, s, radix) = 0;
    return s;
}
```

cnlohr, 2023-06-26:
Normally that comes with libgcc, but, I appreciate the answer - do you mind if we close the issue?

karstengit, 2023-06-26:
Addtionally installed package libgcc-10-dev-riscv64-cross and dependent packages with many MB data.

No change - it still does not find an ultoa or sprintf.
I have no idea where to search?

Maybe an newer version of gcc-riscv64-unknown-elf will help, but it is not easy to get it into the stable system.

mengstr, 2023-06-26:
A version of sprint() is included in the fun-library, but it's not in the .h file so you need to declare it yourself to avoid warnings.

```
int mini_snprintf(char* buffer, unsigned int buffer_len, const char *fmt, ...);

int main() {
  char buf[20];
  mini_snprintf(buf, sizeof(buf), "%s %x", "Hello", 10);
}
```

karstengit, 2023-06-27:
> The Makefile explicitly builds with `-nostdlib`, maybe that's the problem?

Yes - I already did see this too and tried to compile without this option, but there where other problems I cannot remember.
 
> In the MounRiver toolchain I can build with stdlib functions perfectly fine, but I also add `--specs=nano.specs --specs=nosys.specs`, which doesn't work with the standard GCC because they don't seem to have those libs precompiled for rv32ec

O.K. Thanks. The meaning is not clear to me at this time, but it is a good hint.

I will try to setup a small Debian sid (unstable) with a newer gcc-riscv64-unknown-elf 12.2.0-14 first and see how the behaviour will be there.

karstengit, 2023-06-27:
System upgrade done. The positive message is that this part works without problems.

The negative message is that the upgrade does not effect the described problem of the bug.

```
$ make
riscv64-unknown-elf-gcc -o spi_lcd.elf ../../ch32v003fun/ch32v003fun.c spi_lcd.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
spi_lcd.c: In function 'main':
spi_lcd.c:93:9: warning: implicit declaration of function 'ultoa'; did you mean 'utoa'? [-Wimplicit-function-declaration]
   93 |         ultoa(SYSTEM_CORE_CLOCK / 1000000, lpuffer, 10);
      |         ^~~~~
      |         utoa
/usr/lib/gcc/riscv64-unknown-elf/12.2.0/../../../riscv64-unknown-elf/bin/ld: /tmp/ccPVLYzm.ltrans0.ltrans.o: in function `.L0 ':
/win/WCH/K/examples/spi_lcd/lcd.c:110: undefined reference to `ultoa'
/usr/lib/gcc/riscv64-unknown-elf/12.2.0/../../../riscv64-unknown-elf/bin/ld: /win/WCH/K/examples/spi_lcd/lcd.c:167: undefined reference to `ultoa'
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:27: spi_lcd.elf] Fehler 1
```
**undefined reference to `ultoa'**

```
$ make
riscv64-unknown-elf-gcc -o spi_lcd.elf ../../ch32v003fun/ch32v003fun.c spi_lcd.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
/usr/lib/gcc/riscv64-unknown-elf/12.2.0/../../../riscv64-unknown-elf/bin/ld: /tmp/cc98ACmE.ltrans0.ltrans.o: in function `.L0 ':
/win/WCH/K/examples/spi_lcd/lcd.c:116: undefined reference to `sprintf'
/usr/lib/gcc/riscv64-unknown-elf/12.2.0/../../../riscv64-unknown-elf/bin/ld: /win/WCH/K/examples/spi_lcd/lcd.c:166: undefined reference to `sprintf'
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:27: spi_lcd.elf] Fehler 1
kars
```
**undefined reference to `sprintf'**

cnlohr, 2023-12-24:
I'm going to mark this as complete now.

closed 2023-12-24 by cnlohr

TommyMurphyTM1234, 2024-01-08:
> In the MounRiver toolchain I can build with stdlib functions perfectly fine, but I also add `--specs=nano.specs --specs=nosys.specs`, which doesn't work with the standard GCC because they don't seem to have those libs precompiled for rv32ec

Poking an old (closed) issue but, in case it helps anyone, the xPack GNU RISC-V Embedded GCC toolchain bundles Newlib nano support for `rv32ec/ilp32`:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#releases
    * https://xpack.github.io/blog/2023/09/25/riscv-none-elf-gcc-v13-2-0-2-released/