cnlohr/ch32fun #199: Move libgcc.a?

mengstr, 2023-06-29:
Can we move `libgcc.a` from `/misc/` into `/ch32v00fun/`?  All the examples here are using that lib and so does my own Makefiles since they are loosely based on the `Makefile.mk` from here.

It would be nice to have all the fun-files that are required for a project contained in one single source folder.

I have a target that looks like this in my Makefiles pulling in the latest version of this project and removes all superfluous files and directories leaving only what's needed.  Sure it works by having it collect the bits and pieces from various folders, but it feels not ideal.

```
getfun :
	rm -rf ch32v003fun tmpfun
	git clone --depth 1 https://github.com/cnlohr/ch32v003fun.git tmpfun
	cp tmpfun/misc/libgcc.a tmpfun/ch32v003fun
	mv tmpfun/ch32v003fun/ . 
	rm -rf tmpfun
```

cnlohr, 2023-06-29:
The con: Ideally libgcc.a should be optional, and you can use the system libgcc.a, additionally, it seems others want to include support for other ch32vXXX's, and that libgcc.a doesn't use multiply/divide.  Additionally, libgcc.a is very big (storage-wise) compared to the other ch32v003fun files.

The pro: That libgcc.a is really good for ch32v003's.  It also fixes most of the problems people have with their toolchains.   It would be nice for it to be part of everything that gets pulled in.

Neutral: Does this only cause annoyance when needing to copy out a few files?

ALSO - Do you have any good recommendations for how people should "use ch32v003 in their projects"?  Is it already too big for a submodule?  Should we sub-module *just* ch32v003fun* and include it in an outside thing?

I will consider this - if anyone else has opinions they should chime in.

mengstr, 2023-06-29:
Regarding the con it would then seem logical that the Makefile.mk (for the examples) would use the system installed libgcc.a instead of the one supplied by fun. 

Many users probably just copies blink or another example and use as a base for their own projects thus necessitating either hackery of the makefile or implementing a directory structure with the `misc` directory in their project.

cnlohr, 2023-07-06:
Could we just include misc in -L, and use -lgcc?  That way if it exists, then it uses it, otherwise it tries to use the system libgcc?

mengstr, 2023-07-07:
That's a reasonable middle ground. And an interesting one at that.  But are he two libs so compatible that the users code would never never break or increase in size if the misc/libgcc was removed and the toolchain started to use the system one?

But as said - it feels like a bad pattern to pull in code/data from multiple locations for such small project (at least the code that actually is used in the users project is).  

And the optimised libgcc is a small file that can't be rebuilt by the Makefile and thus should be considered like any other asset in a repo.

cnlohr, 2023-07-09:
Actually, we are already doing that it seems.

With libgcc.a from the `misc` folder, and build, I get this:

```
riscv64-unknown-elf-gcc -o template.elf ../../ch32v003fun/ch32v003fun.c template.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
riscv64-unknown-elf-size template.elf
   text	   data	    bss	    dec	    hex	filename
   1756	      0	      4	   1760	    6e0	template.elf
```

If I remove it, I get this:
```
riscv64-unknown-elf-gcc -o template.elf ../../ch32v003fun/ch32v003fun.c template.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections -L../../ch32v003fun/../misc -lgcc
riscv64-unknown-elf-size template.elf
   text	   data	    bss	    dec	    hex	filename
   1816	      0	      4	   1820	    71c	template.elf
```

I think everything here is in order?  Should I close the issue?

It's not a great pattern but it seems like the best compromise to me?

mengstr, 2023-07-09:
ok, good enough for me. :) But It might be nice to mention in the readme that the project supplied libgcc will produce smaller code and it's highly recommended to use it.

mengstr, 2023-07-20:
Well...   I tried this but no joy.
```
mats@Matss-MacBook-Pro blink % make

riscv64-unknown-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -T ../../ch32v003fun/ch32v003fun.ld -Wl,--gc-sections  -lgcc
/opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(div.o)
collect2: error: ld returned 1 exit status
```

cnlohr, 2023-08-01:
It sounds like your libgcc is only for 64-bit systems, you are a prime person who would benefit from using the bundled libgcc.

cnlohr, 2023-08-01:
Looks like you are linking your toolchain's libgcc which is not for 32-bit ilp.

While you are a prime person who would benefit from the bundled libgcc, it isn't a critical part of every individual using ch32v003fun.  It feels weird for something that _isn't_ ch32v003fun to be in the ch32v003fun folder.

I don't think I want to move libgcc.a :-/.

If there is a renewed push for some reason that's fine by me, but at this time, please let's just keep things as they are.

closed 2023-08-01 by cnlohr