cnlohr/ch32fun #207: DMA interrupts mixed up in WS2812B driver

dquadros, 2023-07-11:
In the DMA IRQ handler, Transmission Complete and Halfway Empty tests are switched:

```
if( intfr & DMA1_IT_TC3 )
{
	// Halfwaay (Fill in first part)
	WS2812FillBuffSec( WS2812dmabuff, DMA_BUFFER_LEN / 2, 1 );
}
if( intfr & DMA1_IT_HT3 )
{
	// Complete (Fill in second part)
	WS2812FillBuffSec( WS2812dmabuff + DMA_BUFFER_LEN / 2, DMA_BUFFER_LEN / 2, 0 );
}

```
It should be

```
if( intfr & DMA1_IT_HT3 )
{
	// Halfwaay (Fill in first part)
	WS2812FillBuffSec( WS2812dmabuff, DMA_BUFFER_LEN / 2, 1 );
}
if( intfr & DMA1_IT_TC3 )
{
	// Complete (Fill in second part)
	WS2812FillBuffSec( WS2812dmabuff + DMA_BUFFER_LEN / 2, DMA_BUFFER_LEN / 2, 0 );
}

```

cnlohr, 2023-07-30:
I believe you are right but it makes things not work on my board.  Does flipping them work for you?

dquadros, 2023-07-30:
Yes it worked. I have to find the code and clean it up, but you can see (briefly) my demo here: https://youtu.be/UYo61C7A4T4?t=306

cnlohr, 2023-07-31:
Understood.  I tested and it worked fine. Fixed in #3bd7770

closed 2023-07-31 by cnlohr