cnlohr/ch32fun #219: debugprintfdemo bricks CH32V003J4M6 since SWIO and PD4 share the same PIN

limingjie, 2023-08-01:
# Root Cause
The `SWIO` and `PD4` share the same PIN on CH32V003J4M6. In the `debugprintfdemo`, `PD4` is used for output, after flashing the chip is bricked since `SWIO` is taken as `PD4`.

# Solution
- The `debugprintfdemo` should avoid using `PD4`.
- The command `minichlink -ua` can unbrick the chip. However, if the chip was connected to the WCH-LinkE for a long time, it could be damaged. For me, one chip is unbricked, and another is too hot and damaged.

[image]

cnlohr, 2023-08-02:
PR accepted.

What was WCH thinking when they put those pins together????

closed 2023-08-02 by cnlohr

limingjie, 2023-08-02:
Thank you! I was also wondering why they share the SWIO pin, to easily brick the chip and make more earnings? LOL.

Luckily, the `minichlink` can unbrick the chip, great tool! Thanks again!