cnlohr/ch32fun #225: Solution for using strings in flash instead of RAM

karstengit, 2023-08-08:
This is a question or feature request and not a bug.

I just started to build my first project with a CH32V003F4P6 that uses an 4x10 character LCD display.
So it is needed to handle strings with text of course.
When you are working with much strings you are running out of RAM at a certain point.

Is there a possibility to do something like the **F()** macro in Arduino or or the **PSTR()** macro in the avr-gcc?
At this time I could not find out how this macros have been implemented - only that it has to do with PROGMEM.

Has someone already an solution for it?

mengstr, 2023-08-08:
Isn't it just as simple as making the strings `const` and they will end up in the _text_ segment instead of _data_ ?

`char foo1[]="ABCDEFGHIJABCD";`
vs
`const char foo2[]="ABCDEFGHIJABCD";`

mengstr, 2023-08-08:
The reason for AVR having to do special things to get strings into the flash is because it's a Harvard architecture where the data and code memories are totally separated. There's a special instruction to read from code memory into a register.

The RISCV is Von Neumann where code and data (and in this case also IO/Peripherals) shares a single memory address area. So you don't need to jump through hoops to get the desired RAM savings.