cnlohr/ch32fun #227: WCHLink-E not working with minichlink

jeremyherbert, 2023-08-17:
I have a new WCHLink-E device and it didn't work with minichlink. It turns out that the programmer can enumerate with different PIDs depending on the working mode. For the WCH proprietary mode to work with minichlink, it needs to enumerate as VID:PID 1a86:8010. However, out of the box mine enumerated with 1a86:8012. minichlink gave the error "Could not initialize any supported programmers".

I have since discovered that this is because my devices came shipped with the ARM mode active rather than WCH/RISCV proprietary SWIO:

[image]

If your programmer has the blue LED on by default, this is the problem.

To change the mode, you need to remove the PCB from the enclosure and hold down the button labelled `ModeS` while you plug it in to your USB port (see pink arrow):

[image]

It seems to store the mode in non-volatile memory, so you don't need to hold down the button every time you power the device on; you just need to do it once.

cnlohr, 2024-01-21:
Thank you so much for creating this ticket.  I would have spent forever.

Nable80, 2024-01-21:
FYI: it looks like there's a command to switch the mode: https://github.com/wagiminator/MCU-Flash-Tools/blob/main/rvmode.py

I wonder if the same command can switch it back too.

cnlohr, 2024-02-12:
If I was silly and I pressed and held the IAP button while plugging it in because it was upside-down, does anyone know if there's a way to fix my programmer?

eeucalyptus, 2024-02-12:
Afaik it's just a matter of doing the same thing again

cnlohr, 2024-02-13:
I have added features to minichlink to automate the process to exit IAP and/or switch out of ARM mode.