cnlohr/ch32fun #231: no return for *twoway_strstr() function

eznas, 2023-08-23:
MounRiver reported "static char *twoway_strstr(const unsigned char *h, const unsigned char *n)" no return error/warning, and shall be fixed.

eeucalyptus, 2023-08-23:
Please ask the MounRiver or wch support. ch32v003fun is a community project that aims to reduce dependencies in proprietary software, not support it ;)

TommyMurphyTM1234, 2024-06-18:
I don't get any such error/warning when compiling code that uses `twoway_strstr()` via `strstr()` using:

- [xPack GNU RISC-V Embedded GCC](https://xpack.github.io/dev-tools/riscv-none-elf-gcc/) [v13.2.0-2](https://xpack.github.io/blog/2023/09/25/riscv-none-elf-gcc-v13-2-0-2-released/)
- [MounRiver Studio 1.9.1](http://www.mounriver.com/download) xPack GNU RISC-V Embedded GCC v8.2.0 
- [MounRiver Studio 1.9.1](http://www.mounriver.com/download) xPack GNU RISC-V Embedded GCC v12.2.0

Unless instructions for reproducing the issue can be provided I reckon that this issue could/should be closed.

closed 2024-06-19 by cnlohr