cnlohr/ch32fun #233: standby_autowake problem?

JorjBauer, 2023-08-27:
I don't know if I've got bad silicon, or if there's something wrong with this example - it never gets past the __WFE() call, instead restarting as soon as __WFE() is issued. It sort of looks like it might be sleeping and autowaking because of the simplistic blink - but it doesn't; adding a second LED that turns on after resuming from sleep shows that it never gets that far. And the power consumption never drops anywhere near sleep levels.

eeucalyptus, 2023-08-27:
Bad silicon is normally very unlikely, though of course not impossible. The more likely thing is software. Can you please provide code? Did you already try the standby example?

recallmenot, 2023-08-27:
Might I add - flashing places the ch32v003 in a state that makes sleeping impossible until next boot from power-off.

JorjBauer, 2023-08-27:
This is the standby example - I'm just not connected to serial, so I'm relying on LEDs to show whether or not it's working. I just commented out the printfs, and it's clear that it's rebooting and not waiting for the AWU timer.

The comment about sleep after flash is interesting, I'll check that out - it doesn't match my experience so far, though. I've gotten low power states for wake-for-interrupt without power cycling after flash. Does that apply to certain sleep modes? And is it documented somewhere?

JorjBauer, 2023-08-27:
LED on PD4, Serial on PD5, defined #define FUNCONF_USE_UARTPRINTF 1 in funconf.h - and it emits loops of "5000ms wait over" and "entering sleep loop" but never emits "awake, <counter>". Straight stock code from standby_autowake.

JorjBauer, 2023-09-01:
The problem seems to be that there's an external crystal on A1/A2 on the dev board I was using (still using the internal oscillator, but the crystal is there). Retested on a bare chip and it works as expected. I'm not sure why it won't stay asleep with an external crystal.

JorjBauer, 2023-09-02:
And to follow up on sleep-after-flash, for anyone looking at the same problems -- I've reproduced that. The ch32v003 enters sleep just fine (down to about 1.5mA), but won't enter standby, aka deep sleep (best I've gotten is about 20µA) without a cold boot after being flashed. In that state the debug pin is ignored, unless you've also set a wake event for port D1 to rouse it from standby...

cnlohr, 2023-09-04:
> And to follow up on sleep-after-flash, for anyone looking at the same problems -- I've reproduced that. The ch32v003 enters sleep just fine (down to about 1.5mA), but won't enter standby, aka deep [...]

This is EXTREMELY INTERESTING and should be documented.  Can I copy/paste your reply into the comment in the code?  Do you mind making a PR to outline this in more detail?

cnlohr, 2023-09-04:
I am adding the following to the example: 
**WARNING: You MUST hard-reboot the CH32V003 to allow it to go into deep sleep. You cannot go from flashing to deep sleep without a hard power cycle.**

cnlohr, 2023-09-04:
It also makes sense that during deep sleep it would not respond.  This is definitely territory where we would expect a need for `unbricking` to fix it if stuck in deep sleep.

JorjBauer, 2023-09-06:
Fine to use my words if you like. I'm still investigating this - I agree that it's very interesting. @recallmenot do you know where you found out that sleep and flash were related? I still haven't seen any documentation about this at all. I'm hoping to find out that it's because the debug interface is still active in some way, and that there's a way to deactivate it without a full power cycle...

JorjBauer, 2023-09-06:
@cnlohr the D1 interrupt does work correctly when in deep sleep, though. Might be worth adding a warning about that, too -

`  // D1, too - the debugging/programming line, which we need to wake            
  // so we can get back in to programming mode                                  
  AFIO->EXTICR |= (uint32_t)(EXTI_PORTD << (2 * 1));
  EXTI->EVENR |= EXTI_Line1;
  EXTI->FTENR |= EXTI_Line1;
`

... with the necessary clock enables for AFIO and external interrupt code of course.

JorjBauer, 2023-09-06:
Found it - https://www.wch-ic.com/downloads/file/369.html - page 19: "When the microprocessor is in Debug mode, it is not possible to enter any kind of Sleep mode."

Experimentally I find that not to be true (light sleep worked in my tests, as I said). But it's good to find confirmation in the docs. Now we just need to figure out how to get it to exit debug mode. Maybe writing 0 to dmcontrol's dmactive bit. I guess I'm about to build a programmer to test that theory...

JorjBauer, 2023-09-11:
I think this is the fault of the programming device (and/or lack of proper documentation), but haven't fully proved it yet.

I built a simplistic programmer on a Teensy 3.6 over the weekend and got to the point where it reads flash (no writing capability) and put a simplistic sleep test program on a CH32v003 which - lights a LED for a second; then turns it off and sleeps until a timer expires in 30s, a button is pressed, or the debugger pin interrupt fires; moves on to the next LED on the board; and repeats. I get good periods of high current use (full processor speed and LED on, about 10mA) and visible sleep times (at about 150µA) and it's easy to identify if the chip reboots.

If the debugger halts the processor, reads flash, and resumes the processor then power consumption during sleep goes up to about 400µA. Dropping the debugging session or sending a reset doesn't change that behavior - but if I add a deliberate set of dmcontrol's dmactive bit to 0, the power usage goes back to the original 150µA.

commit e9896a2d (2025-02-09): Closes #233, adding comments from issue. I could not repro the solution, but I kept it in the notes.