cnlohr/ch32fun #234: blink bricked CH32V003A4M6?

xsrf, 2023-09-02:
Hey,
so is it possible that your blink example basically bricked my `CH32V003A4M6`? 🙈 
The `CH32V003A4M6` has `SWIO` and `PD4` on the same physical pin. Your blink example toggles `PD4` instantly. I use the official WCH-Link to flash the chip but after flashing blink (I've successfully flashed other code too) it won't connect anymore.
```
> minichlink.exe
Found WCH Link
WCH Programmer is LinkE version 2.8
link error, nothing connected to linker
Could not setup interface.
```
Do you know of a way to recover it?

dwillmore, 2023-09-02:
Use the "-u" option of mminichlink.  You'll need to have the WCH-LinkE
control the power to your board.

xsrf, 2023-09-02:
@dwillmore thx, that helped ;) 
I've seen the `-tf` option to disable the power supply and hoped it would work when it just enables the power for flashing, but that was not the case. Missed the `-u` somehow - thx!

Maybe adding a delay to the example at the beginning would help with that kind of issue? :)

dwillmore, 2023-09-02:
It may.  I've been meaning to make some changes to the blink anyway.