cnlohr/ch32fun #243: curiosity

cyrax50000, 2023-09-24:
hello,
while I applaud your minimalism, I am wondering what are you actually doing with the devices? At least with the esp8266, it has internet so I can try to make all kinds of clients and servers with the sdk, however with an offline device I don't have so many ideas ! maybe i'll buy a few of these too if I find some good uses. keep up the good work :)

cnlohr, 2023-09-24:
you can do all sorts of things.  I'll just list some random ideas that they are well suited for.

Turning an analog potentiometer, encoder, button, LED, etc.  into I2C devices. 
Adding USB interfaces to an analog input.
Making touch buttons that can be SPI or I2C, or even USB.
Using a TIA to detect tiny photocurrents and demodulate signals.
Building very simple synthesizers.

cnlohr, 2023-09-24:
Among all the other projects others are doing!

cyrax50000, 2023-09-24:
Wow, thanks for the quick reply. I couldnt understand much of what you said, so i'll go read about these things, for now i'll keep playing with the esp 01, good luck with the projects ! !

nolanhergert, 2023-09-25:
However, all of those things cnlohr listed are things that the ESP series can *also* do (broadly).

If you don't *need* wifi though, the CH32V003 is 1/5 the cost and 1/3 the size (if you use the QFN package) of a minimal ESP8266 chip. https://jlcpcb.com/partdetail/EspressifSystems-ESP8266EX/C77967

It's not a big deal for prototypes, but it's limiting when you want to make more than 10 of something for normal folks who care about things like cost and aesthetics.

As for ideas, there are a lot of single-application, non-"smart", cost-focused, and hobbyist accessible products out there. Maybe give Kickstarter or Hackaday.io a browse? I find dollar stores inspiring too.

mengstr, 2023-09-25:
I have the same question about what everybody are using their Raspberry Pi's for. Except for a "digital signage" (A TV that displays information) I've never had any use for them at all.

Back to the CH32V003. I've recently made a KIM-1 clone (where the CH32 is emulating the 6502 processor). Also a Clock & Reset module for breadboards. And a small 7-segment  module that decodes 12 bits of binary input and displays it on three displays as hexadecimal.

The CH32 is cheaper than most simple logic ICs like the popular 74HC595 that are used in many projects and ca easily replace it. It it in the price range of a simple 555-timer chip, but with much more flexibility.

Not everything needs to be a IoT device even if it seems like that would be the case sometimes. ;-)

eeucalyptus, 2023-09-25:
One of the main usages of the ch32v003 is for me is as a timer. I know that people tend to be like "why not use a 555", but MCUs have some serious advantages like precision, number of passives, long time delays and others. And now that the price of the 003 is lower than the price of a timer, I see no reason to use timer ICs anymore :D