cnlohr/ch32fun #246: minichlink debugger instability

lzace817, 2023-10-03:
The debugger with minichlink is quite unstable.

But instability is subjective, I would like to provide something concrete about it.

In the wireshark dump bellow at line `00000427` should be:
- 'OK'     for success
- 'E NN'   for an error

```
0000022E  24 6d 31 31 36 2c 34 23  36 35                     $m116,4# 65

0000041A  2b                                                 +
0000041B  24 37 33 30 30 32 30 33  30 23 38 66               $7300203 0#8f

00000238  2b                                                 +
00000239  24 44 23 34 34                                     $D#44

00000427  2b                                                 +

0000023E  2d                                                 -
0000023F  2d                                                 -
00000240  2d                                                 -
00000241  2b                                                 +
```
> Note: `RVHandleDisconnect` seems to be the one handling it.

I hope it is usefull.
## References
[manual](https://sourceware.org/gdb/current/onlinedocs/gdb.pdf) page 774

lzace817, 2023-10-06:
freedom.

AFAIK wchlink doesn't work with vanilla openocd. Given the modified version of openocd and the FOSS software nature, perhaps it's already supposed to be open, but I'm not a lawer. But having more control over our tooling it good. If anyone knows better, I would be glad to hear.

lzace817, 2023-10-06:
At this moment, minichlink is pretty much unusable for debugging. It's ok for flashing and serial i/o.

About your specific problem, I'm wondering if you can drop the `verify`, because it may be the source of slowdown.  
No reflash needed. On windows I only used the MRS, don't know about drivers.

LaXiS96, 2024-11-22:
I am not able to meaningfully use the minichlink debugger with VSCode, stepping is pretty random and continuing doesn't do anything.
I was wondering if it was possible to use OpenOCD for debugging while using minichlink for debug output with the custom printf implementation over SDI... I guess not, since I think SDI runs through the same interface as the main debugger.

cnlohr, 2024-11-22:
1. @LaXiS96 Please use the gdb_fix branch.  This fixes almost all cases, except for stepping "over" functions that take a really long time, like delays.
2. @mrx23dot I am desperate to find one last corer case that I cannot figure out with gdb.  Is there any chance you could be bothered to try it out as it stands in the gdb_fix branch?

cnlohr, 2024-11-28:
Well, some of this feedback I am incorporating to the feedback to go into the next round of changes to minichlink, but for, gdbserver there's

```
make gdbserver
```
and
```
make gdbclient
```

We definitely can improve the documentation.

Also, I had never seen the wch_swio_flasher.  It's unfortunate it uses emulated NCH link protocol, because it's so hard to materially improve that.

Also, I too have noticed that recent versions seem to burn slower.  What device were you using to flash when writing 8kB took 27 seconds?!!?   It should take about 1.6 seconds to flash 8kB theoretically.

cnlohr, 2024-11-28:
RST is generally not used when programming the 003.  Only SWIO.

Blue flashing _may_ mean that it's in bootloader mode.  Minichlink "should" automatically exit bootloader mode if that's the case.

Can you repost the full output, of your command, and the output of minichlink?

cnlohr, 2024-11-29:
There is no binary. 

https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.mk#L212-L213

```
gdbserver : 
	-$(MINICHLINK)/minichlink -baG
```

We are also in progress to improving flash performance on Windows.