cnlohr/ch32fun #248: [Question] CH32 as I²C device with display also I²C

GSoftwareDevelopment, 2023-10-07:
Hi.

I wanted to use the µC CH32 as an I²C slave device to convert simple messages into graphical data and display it on an OLED screen also connected to I²C.

I've been browsing the resources of this repository and I see that there are nice examples of I²C-Slave and OLED I²C support, however, a question is bothering me. Using both examples on the same chip, won't they interfere with each other?

Both examples use `I2C1` - I understand that this is an "object" representing the I²C bus on the µC CH32. Both examples configure `I2C1` in their own way - similar, but yet with a different purpose (one as a Slave and the other as a Master) Both examples use an IRQ (presumably the same?).

My question is, do the above statements translate into reality? Is it possible to use the I²C bus as master and slave at the same time, so that a device can receive data from another µC via the I²C bus and control a display embedded also on the I²C bus?

I would appreciate any info on this :)

eeucalyptus, 2023-10-07:
It is absolutely possible (though uncommon) to use nodes in both roles. You would just have to keep it in slave mode until you decide to use it as master. Then you Just re-initialize the module as master until that transfer is done. You have to take precaution on bus arbitration because of the multi-master bus. 

About just "combining" the examples, it's of course a bit more than just pasting one example into the other, but I guess you will get quite far with the existing code

GSoftwareDevelopment, 2023-10-08:
OK, so in general I need to initialise the µC as a slave (example "i2c-slave") and listen for incoming messages. In case the µC/Slave needs to perform an operation on the display, I switch the µC to master mode (example "i2c-oled") perform the operation, then restore (re-initialise) Slave mode. Generally, this does not seem complicated :)

However, I am bothered by the arbitration issue. At first, I didn't really understand what it was about, but I found an explanation of the phenomenon. I don't know if I'll be able to deal with it. The control device (master) can send a change message or a request for information at any time. The examples provided, do not seem to solve this type of dispute.

Well... will give it a try. We'll see how it works without implementing arbitration. :)

eeucalyptus, 2023-10-09:
Good luck with it, consider sharing your findings about it in the discord channel (see project readme)