cnlohr/ch32fun #252: how do I compile minichlink on a mac?

thirstyone, 2023-11-01:


mengstr, 2023-11-01:
Like this: https://github.com/cnlohr/ch32v003fun/wiki/Installation#macos

So basically:
1) Install the toolchain and prerequisites as in the wiki
2) `cd` to the folder with minichlink
3)  Type `make` and the executable will be built

I've done this a few times on my Mac

thirstyone, 2023-11-01:
I haven't seen those installation instrcutions, but I noticed a makefile inside the minichlink folder, so of course I tried it, and here's what I got: 

> gcc -o minichlink minichlink.c pgm-wch-linke.c pgm-esp32s2-ch32xx.c nhc-link042.c ardulink.c serial_dev.c pgm-b003fun.c minichgdb.c -lpthread -lusb-1.0 -framework CoreFoundation -framework IOKit -O0 -Wall -Wno-asm-operand-widths -Wno-deprecated-declarations -Wno-deprecated-non-prototype -D__MACOSX__  -I. -I/usr/local/Cellar/libusb/1.0.26/include/libusb-1.0 -L/usr/local/Cellar/libusb/1.0.26/lib
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
1 warning generated.
warning: unknown warning option '-Wno-deprecated-non-prototype' [-Wunknown-warning-option]
In file included from minichgdb.c:26:
./microgdbstub.h:735:70: error: use of undeclared identifier 'MSG_NOSIGNAL'
                        ssize_t rx = recv( serverSocket, (char*)buffer, sizeof( buffer ), MSG_NOSIGNAL );
                                                                                          ^
./microgdbstub.h:793:34: error: use of undeclared identifier 'MSG_NOSIGNAL'
                send( serverSocket, data, len, MSG_NOSIGNAL );
                                               ^
1 warning and 2 errors generated.
make: *** [minichlink] Error 1
192:minichlink red$ make config
make: *** No rule to make target `config'.  Stop.

However, I obtained the toolchain from the MounRiverStudio website, not through homebrew...

thirstyone, 2023-11-01:
heh. I moved that define outside the if, and it compiled! 
Then I run this line 

> ./minichlink -a -w /Users/red/Documents/dev/ch003/CH32V003-makefile-example-main/build/example.hex 0x08000000 -T

and got the following output: 

> Found WCH Link
> WCH Programmer is LinkE version 2.9
> Chip Type: 003
> Setup success
> Flash Storage: 16 kB
> Part UUID    : 84-22-ab-cd-ec-46-bc-57
> PFlags       : ff-ff-ff-ff
> Part Type (B): 07-13-bb-91
> Read protection: disabled
> Interface Setup
> Fault writing memory (DMABSTRACTS = 08000302) (Execption executing Abstract Command) DMSTATUS: 004c0382
> Fault on DefaultWriteWord Part 1
> Fault writing memory (DMABSTRACTS = 08000302) (Execption executing Abstract Command) DMSTATUS: 004c0382
> Fault on DefaultWriteWord Part 2

with multiple 'fault' lines. I removed -a flag, removed -T flag, but still the same error.  And it seems, it corrupts the firmware flashed onto the chip, so that I have to reflash it with openocd to make it work again.

public-radiation, 2023-12-24:
Have you tried using -p to remove the read protection? It can be set by firmware (optionbytes) or by -P.

(i had the same issue, and that brought me to this issue/question)

cnlohr, 2023-12-24:
> I haven't seen those installation instrcutions, but I noticed a makefile inside the minichlink folder, so of course I tried it, and here's what I got:
[snip]
> > make: *** [minichlink] Error 1
> > 192:minichlink red$ make config
> > make: *** No rule to make target `config'.  Stop.
> 
> However, I obtained the toolchain from the MounRiverStudio website, not through homebrew...

I take it you resolved your issue @thirstyone ? What needed to change? I can't imagine any sane mac build not having MSG_NOSIGNAL

monte-monte, 2025-11-25:
I followed installation instructions in the wiki and was able to successfully build on mac.