cnlohr/ch32fun #255: Duplicate readings in `handle_debug_input()` and more..

arduino12, 2023-11-23:
**Hi,**

I finally got bunch of chips and started playing - I love this repo so much!
Thanks for this and the videos! @cnlohr and other contributors !

I try to use the debug input feature with the `printf()` so I can make an interactive command menu.
(will surly add an example PR once its working!).

I used the [debugprintfdemo](https://github.com/cnlohr/ch32v003fun/blob/master/examples/debugprintfdemo/debugprintfdemo.c) but I found some weird behavior (I use Windows 10 VS code + PIO with ESP32 S2 Mini as a programmer):
1. The example didn't work at first with the PIO tool-minichlink package-
(`cd ~/.platformio/packages/tool-minichlink` -> `.\minichlink.exe -bT` in VS code terminal)
I **[opend an issue](https://github.com/Community-PIO-CH32V/tool-minichlink/issues/1)** to update the `minichlink.exe` and now it works.

2. I found out that the `poll_input()` is not working, only the print functions resulted in `handle_debug_input()` being called.
So if the `debugprintfdemo` was not calling the `printf()` it wouldn't actually receive the debug input..
I find out that calling `putchar('\0')` gives the desired polling effect so I use it for now every 100ms (but seems wasteful).

3. After saving the debug input to a global "stdin" buffer, I found characters are being duplicated an even number of times,
Like when I send "A" I may get "AA" or "AAAA", for "ABC" I may get "AABBCC" or "ABABCC"..
I solved it by [removing these lines from `putchar()`](https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c#L979-L984) - but I don't fully understand the state machine there...

4. I think it will be super nice if the input API was like the output API,
like using `scanf()` or `getchar()` but I don't know how to do it so I may use `atoi()` or `sscanf()` for now..

**Here is my code:** (See "3" above - it will echo duplicated characters unless the `putchar()` hack fix is applied)
```
#include "ch32v003fun.h"
#include <stdio.h>
#include <string.h>

#define CMD_BUFFER_SIZE			(64)
#define CMD_POLL_PERIOD_MS		(100)

char cmd_buffer[CMD_BUFFER_SIZE];
uint8_t cmd_buffer_index = 0;

/* Callback from poll_input() and putchar() - not an ISR but can't block here too */
void handle_debug_input(int size, uint8_t *buffer)
{
	uint8_t cmd_ready = 0;

	while (size--) {
		/* Treate a new-line character as the end of the command  */
		if ((*buffer == '\r') || (*buffer == '\n')) {
			cmd_ready = cmd_buffer_index;
		}
		else {
			cmd_buffer[cmd_buffer_index++] = *buffer++;
			/* Treate a full buffer as the end of the command  */
			if (cmd_buffer_index >= CMD_BUFFER_SIZE - 1)
				cmd_ready = 1;
		}
		/* Signal main function a new command is readable from the buffer */
		if (cmd_ready) {
			cmd_buffer[cmd_buffer_index] = '\0';
			cmd_buffer_index = CMD_BUFFER_SIZE;
			/* Ignore the rest of the input - main will now handle the command buffer */
			return;
		}
	}
}

int main()
{
	SystemInit();

	SysTick->CNT = 0;
	uint32_t cur_time = SysTick->CNT;
	uint32_t next_putchar_time = 0;

	printf("Enter command and press ENTER:\n");
	while(1)
	{
		cur_time = SysTick->CNT;

		/* Poll debug input every CMD_POLL_PERIOD_MS */
		if (((int32_t)(cur_time - next_putchar_time)) > 0) {
			next_putchar_time = cur_time + Ticks_from_Ms(CMD_POLL_PERIOD_MS);
			/* poll_input() not working, so I find this hack */
			/* Proccess debug line DMA buffer and calls handle_debug_input() when needed */
			putchar('\0');
		}

		/* Did we got a full command? (debug input that ends with a new-line) */
		if (cmd_buffer_index == CMD_BUFFER_SIZE) {
			printf("> [%02d] %s\n", strlen(cmd_buffer), cmd_buffer);
			*cmd_buffer = '\0';
			cmd_buffer_index = 0;
		}
	}
}
```

cnlohr, 2023-11-23:
I will actually be at a win 10 machine tomorrow.  

I am very surprised poll_input does not work.  Are you calling it in a loop?  (That's what you should do).

I struggled to get vs code working hardly at all with this, so congrats on that!  Hopefully something can be done to make it more reliable to setup.

hexeguitar, 2023-11-23:
I have experienced it on Linux, too. Reported here on discord:
https://discord.com/channels/665433554787893289/1080242396736000100/1138030258965184544
Interestingly, it happens when using the ESP32-S2 programmer only. WCHLink-E works fine.

arduino12, 2023-11-23:
**Thanks!**

cnlohr,
Yes I firstly use the `poll_input()` just like the original example in a loop and all,
But later I found out that it is not work on its own and the MCU have to print some char in order to get chars too.
And when printing the `handle_debug_input()` is called without needing the `poll_input()` so I didn't use it.

I think we better add comment to the example:
1. The size of the DMA buffer.
(I didn't find how many "stdin" characters it can hold before polling them).
2. A note about `handle_debug_input()` - that the `int size` argument can only be 1, 2 or 3.
(because of the `uint32_t` DMA item size).

For me the VS code + PlatformIO worked right out of the box,
I used your [platformio.ini](https://github.com/cnlohr/ch32v003fun/blob/master/platformio.ini) file and only added this:
```
[env]
board = genericCH32V003F4U6
upload_protocol = minichlink # minichlink, wch-link, isp
```
Finding out about the old version of `minichlink.exe` package took me most of the time.. 😅

hexeguitar,
The discord link shows an empty server to me :(

I will get some WCHLink-E soon so I can try with all of them and report if needed.

I did change the original ESP32 S2 FW a bit, but if it happend to others I am calm it is not related to my small change:
(added another [pinout](https://github.com/cnlohr/esp32s2-cookbook/blob/master/ch32v003programmer/main/pindefs.h) so the 3 pins will be close for a 3P JST connector):
[image]

arduino12, 2023-11-24:
I found a **[WCH-LinkE-R0-1v2](https://www.aliexpress.com/item/1005004964197577.html)**, flashed it with the newest firmware using [WCH-LinkUtility V1.90](https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html),
(apparently pressing the "IAP" button while powering it cause it to erase its flash...).

I connected it to a CH32V003 flashed with my code above and it still shows duplicates characters!
The same fix I found ([removing these lines from putchar()](https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c#L979-L984)) works for it as well.

**It only worked for me using `minichlink -bT` in WSL**,
**I didn't understand how can I see the debug terminal of the WCH programmer under Windows?**
1. `minichlink.exe -bT` outputs:
```
Error: couldn't open wch link device (libusb_open() = -5)
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
```
2. `.\wlink.exe sdi-print enable` from [here](https://github.com/ch32-rs/wlink) outputs:
```
Error: USB error: Entity not found
Caused by:
    Entity not found
```
3. The official WCH-LinkUtility V1.90 connects to it but I didn't find any terminal view...
I tried Target -> Enable SDI printf, and opened the virtual COM port but it didn't show any characters from the chip..
(Also not with the original [debugprintfdemo](https://github.com/cnlohr/ch32v003fun/blob/master/examples/debugprintfdemo/debugprintfdemo.c).. sometimes got this message appeared trimmed:
"Please check the SDI, and Enable this function through the upper computer software")

Once I get the [WCH-LinkE-R0-1v3](https://www.aliexpress.com/item/1005006112031563.html) and the [WCH-LinkW](https://www.aliexpress.com/item/1005006177751678.html) I will test them as well.
**For now the bug seems to be in both Windows and WSL, and in ESP32-S2-Mini and WCH-LinkE-R0-1v2 programmers.**

arduino12, 2023-12-10:
**Hi,**

I got the [WCH-LinkE-R0-1v3](https://www.aliexpress.com/item/1005006112031563.html) flashed it with the newest firmware using [WCH-LinkUtility V1.90](https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html),
and the bug remains - my quick fix also works for it so **I am pretty sure its not related to the programmer.**

Also I understood why last time, the WCH-LinkE-R0-1v2 didn't work with `minichlink.exe` under Windows-
The defaults driver for it and for the WCH-LinkE-R0-1v3 is `WCH-LinkRV`-
So it needs to be replaced to `LibUSB` (using [**USB Driver Tool**](https://visualgdb.com/UsbDriverTool/)) in order for the  `minichlink.exe` to use these programmers.
[image]

Any help on fixing the bug in [ch32v003fun.c](https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c#L979-L984) will be appreciated!

cnlohr, 2023-12-24:
I am finally able to repro this. I am digging in.

cnlohr, 2023-12-24:
I am really sorry for the buggyness here. 0x00 is also a valid byte 0 for no pending transactions. I don't know why polling with setting to 0x8x will cause it to retrigger, and I can't find the code, but there is an easy work around.

cnlohr, 2023-12-24:
polling works, too.

cnlohr, 2023-12-24:
Thank you for doing all this work, @arduino12

commit 157f092e (2023-12-24): Fix #255 - Force the DMDATA0 to 0x00 after successfully reading data out in a poll.
closed 2023-12-24 by cnlohr