cnlohr/ch32fun #262: Reverse engineering the factory bootloader

jrudolph, 2023-12-29:
Did anyone find any documentation of the factory bootloader? Maybe in the chinese documentation?

I started reverse engineering the bootloader here: https://gist.github.com/jrudolph/8c97727c87e03ae645b87297ac348949

cnlohr, 2023-12-30:
It's all over serial, so it's just never been something of interest to me, when there's a fully featured debug + flash + printf interface sitting right next to it that uses only a single wire.  Will leave issue here in case anyone else wants to poke their head in.

jrudolph, 2023-12-30:
Good point, I'm also doing this more for educational reasons :)

Reading the chinese WCH forum, one main point is that the CH32V003 does not have a dedicated pin to choose the boot method, so even if it would initially boot from the bootloader, you would basically only have a single try to flash the user code over UART after which you would have to use the SWIO anyway for further flashing.

Btw. the USART_IAP example in the official EVT zip source code looks very similar to the binary bootloader code but probably not exactly the same (e.g. only 4 functions implemented in source code, while there is a table of 0-7 in the binary, also the watchdog usages seem to be missing in the source code etc).

hubshi, 2024-01-05:
> Good point, I'm also doing this more for educational reasons :)
>
> Reading the chinese WCH forum, one main point is that the CH32V003 does not have a dedicated pin to choose the boot method, so even [...]

Can you please share the original bootloader bin file, i would like to flash it on my board. :)

jrudolph, 2024-01-05:
It's in the attic directory of this repo.