cnlohr/ch32fun #264: How to printf a float number?

eznas, 2024-01-08:
When I use " printf("%f\r\n", 1/3); " , I got only " f "?
How can I print a float number?

emeb, 2024-01-08:
ch32v003fun provides a stripped-down version of printf() which doesn't support floating point numbers. Use of floating point on the ch32v003 MCU is discouraged due to the heavy code size requirements. If you require floating point output you will need to provide that yourself.

eznas, 2024-01-08:
Okay, understood and thanks!

TommyMurphyTM1234, 2024-01-08:
If, having heeded @emeb's sound advice, you still really need `float` support, then maybe consider using a self-contained "tiny" `printf()` implementation, ideally one that doesn't use heap which should generally be avoided in a resource constrained embedded environment. There are several such implementations available. For example;

* https://github.com/mpaland/printf
* https://github.com/eyalroz/printf
* https://github.com/cjlano/tinyprintf
* https://www.menie.org/georges/embedded/small_printf_source_code.html

maxgerhardt, 2024-01-08:
WCH's properiatary toolchain does provide a `printfloat` library that can do that, and that works just fine: https://github.com/Community-PIO-CH32V/platform-ch32v/issues/41#issuecomment-1792337029

However, that is so big in code size anyways that you should never ever use it. Avoid floating pointing at all costs, or do cheaper [fixed-point arithmetic](https://www.embedded.com/fixed-point-math-in-c/) instead

closed 2024-02-02 by cnlohr