cnlohr/ch32fun #265: Why no `-fdata-sections`?

TommyMurphyTM1234, 2024-01-08:
Just curious - is there any specific reason that you don't use `-fdata-sections` along with `-ffunction-sections`:

* https://github.com/cnlohr/ch32v003fun/blob/7e667426ade44b3bbaa6ca542680c31139397c48/ch32v003fun/ch32v003fun.mk#L8

and, ultimately, `-gc-sections`: 

* https://github.com/cnlohr/ch32v003fun/blob/7e667426ade44b3bbaa6ca542680c31139397c48/ch32v003fun/ch32v003fun.mk#L20

to further optimise linked executables for size?

Are there, perhaps, some "magic" data sections that would be impacted by this?

* http://elinux.org/images/2/2d/ELC2010-gc-sections_Denys_Vlasenko.pdf

By the way, thanks for this work. Very interesting. 👍

cnlohr, 2024-01-29:
If you make a PR for this, I will accept it.

TommyMurphyTM1234, 2024-01-30:
> If you make a PR for this, I will accept it.

Thanks @cnlohr - I'll have a look and submit a PR if it looks worthwhile.

TommyMurphyTM1234, 2024-01-30:
Before I prepare a PR, and in case it influences the decision on whether or not to make the change, here are the results of building the examples programs without and with `-fdata-sections`.
* No change in size
```
riscv64-unknown-elf-size blink.elf
   text	   data	    bss	    dec	    hex	filename
    564	      0	      0	    564	    234	blink.elf
    564	      0	      0	    564	    234	blink.elf (-fdata-sections)

riscv64-unknown-elf-size blink_raw.elf
   text	   data	    bss	    dec	    hex	filename
    568	      0	      0	    568	    238	blink_raw.elf
    568	      0	      0	    568	    238	blink_raw.elf (-fdata-sections)

riscv64-unknown-elf-size bootload.elf
   text	   data	    bss	    dec	    hex	filename
    612	      0	      4	    616	    268	bootload.elf
    612	      0	      4	    616	    268	bootload.elf (-fdata-sections)

riscv64-unknown-elf-size cap_touch_adc.elf
   text	   data	    bss	    dec	    hex	filename
   2352	      0	      0	   2352	    930	cap_touch_adc.elf
   2352	      0	      0	   2352	    930	cap_touch_adc.elf (-fdata-sections)

riscv64-unknown-elf-size cpp_virtual_methods.elf
   text	   data	    bss	    dec	    hex	filename
   4184	      4	     12	   4200	   1068	cpp_virtual_methods.elf
   4184	      4	     12	   4200	   1068	cpp_virtual_methods.elf (-fdata-sections)

riscv64-unknown-elf-size direct_gpio.elf
   text	   data	    bss	    dec	    hex	filename
    560	      0	      0	    560	    230	direct_gpio.elf
    560	      0	      0	    560	    230	direct_gpio.elf (-fdata-sections)

riscv64-unknown-elf-size external_crystal.elf
   text	   data	    bss	    dec	    hex	filename
    540	      0	      0	    540	    21c	external_crystal.elf
    540	      0	      0	    540	    21c	external_crystal.elf (-fdata-sections)

riscv64-unknown-elf-size exti_pin_change_isr.elf
   text	   data	    bss	    dec	    hex	filename
    520	      0	      0	    520	    208	exti_pin_change_isr.elf
    520	      0	      0	    520	    208	exti_pin_change_isr.elf (-fdata-sections)

riscv64-unknown-elf-size GPIO.elf
   text	   data	    bss	    dec	    hex	filename
    384	      0	      0	    384	    180	GPIO.elf
    384	      0	      0	    384	    180	GPIO.elf (-fdata-sections)

riscv64-unknown-elf-size iwdg.elf
   text	   data	    bss	    dec	    hex	filename
    660	      0	      0	    660	    294	iwdg.elf
    660	      0	      0	    660	    294	iwdg.elf (-fdata-sections)

riscv64-unknown-elf-size run_from_ram.elf
   text	   data	    bss	    dec	    hex	filename
    424	      0	      4	    428	    1ac	run_from_ram.elf
    424	      0	      4	    428	    1ac	run_from_ram.elf (-fdata-sections)

riscv64-unknown-elf-size spi_24L01_rx.elf
   text	   data	    bss	    dec	    hex	filename
   3780	      8	     24	   3812	    ee4	spi_24L01_rx.elf
   3780	      8	     24	   3812	    ee4	spi_24L01_rx.elf (-fdata-sections)

riscv64-unknown-elf-size spi_max7219.elf
   text	   data	    bss	    dec	    hex	filename
   5808	      0	      0	   5808	   16b0	spi_max7219.elf
   5808	      0	      0	   5808	   16b0	spi_max7219.elf (-fdata-sections)

riscv64-unknown-elf-size standby_btn.elf
   text	   data	    bss	    dec	    hex	filename
[... 145 lines cut ...]
   2240	      0	     72	   2312	    908	input_capture.elf (-fdata-sections)

riscv64-unknown-elf-size tim2_pwm_remap.elf
   text	   data	    bss	    dec	    hex	filename
   2236	      0	      0	   2236	    8bc	tim2_pwm_remap.elf
   2240	      0	      0	   2240	    8c0	tim2_pwm_remap.elf (-fdata-sections)

riscv64-unknown-elf-size spi_24L01_tx.elf
   text	   data	    bss	    dec	    hex	filename
   4120	      8	     24	   4152	   1038	spi_24L01_tx.elf
   4124	      8	     24	   4156	   103c	spi_24L01_tx.elf (-fdata-sections)
```
* Toolchain used (built from the `riscv-gnu-toolchain` [repo](https://github.com/riscv-collab/riscv-gnu-toolchain))
```
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/riscv-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/Downloads/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/Downloads/riscv-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/riscv-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e) 
```
Any opinions @cnlohr  or others?

cnlohr, 2024-02-01:
I have a preference to include -fdata-sections regardless of the marginal increase in size.

closed 2024-02-01 by cnlohr

TommyMurphyTM1234, 2024-02-01:
> I have a preference to include -fdata-sections regardless of the marginal increase in size.

Thanks a lot @cnlohr. 👍