cnlohr/ch32fun #271: Convoluted UART_BRR calculation

ahhuhtal, 2024-01-29:
I was looking through the code to try and pick a snippet for the USART BRR value calculation.

I found the code
```
#define OVER4DIV 4
#define INTEGER_DIVIDER (((25 * (FUNCONF_SYSTEM_CORE_CLOCK)) / ((OVER4DIV) * (UART_BAUD_RATE))))
#define FRACTIONAL_DIVIDER ((INTEGER_DIVIDER)%100)
#define UART_BRR ((((INTEGER_DIVIDER) / 100) << 4) | (((((FRACTIONAL_DIVIDER) * ((OVER4DIV)*4)) + 50)/100)&15))
```

Looking at the reference manual, turns out the USART baudrate register encodes the value `USARTDIV` as a Q12.4 fixed point value. This in turn determines the baudrate as `BAUDRATE = HCLK/(16*USARTDIV)`. But since `USARTDIV` is Q12.4, `16*USARTDIV` is just exactly the baudrate register integer value!

Just using
```
#define UART_BRR ((FUNCONF_SYSTEM_CORE_CLOCK) / (UART_BAUD_RATE))
```
works perfectly well, but rounds sub-optimally in some cases. The rounding can be fixed by
```
#define UART_BRR (((FUNCONF_SYSTEM_CORE_CLOCK) + (UART_BAUD_RATE)/2) / (UART_BAUD_RATE))
```
This always (based on exhaustive checking, while taking CORE_CLOCK = 48 MHz) yields the same or more accurate achieved baudrate as the original expression. Note that the returned register value is not always identical though.

dwillmore, 2024-01-29:
Yes!  I ran into this when I wrote the TX code.  You may also see in the RM
that they base their example on a 72MHz master clock--the whole section was
cut and pasted from some other chips RM--one with a faster clock.  I
thought I was doing the math wrong when I couldn't get the same values as
the #defines got.  Throw in the overly complex way it's calculated in the
code and I wasted way too much time to come to the same realization you
did.  I reported the error to the WCH people but I forgot to make an issue
like you did, so thanks!

cnlohr, 2024-02-01:
I believe this was copied from the EVT.  I agree that this would be a great change.  Would you mind making a PR for this? (Including the rounding fix)

closed 2024-02-12 by cnlohr