cnlohr/ch32fun #272: Redundant `blink[_raw].bin` files source controlled in repo?

TommyMurphyTM1234, 2024-01-31:
Are these an oversight or deliberately source controlled?

* https://github.com/cnlohr/ch32v003fun/blob/master/examples/blink/blink.bin
* https://github.com/cnlohr/ch32v003fun/blob/master/examples/blink_raw/blink_raw.bin

I presume that they should not be there as `make clean` in the relevant example directories deletes them but leaves the cloned repo in this state:
```
git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	deleted:    examples/blink/blink.bin
	deleted:    examples/blink_raw/blink_raw.bin

no changes added to commit (use "git add" and/or "git commit -a")
```
Oddly, `.gitignore` already excludes `*.bin` files:

* https://github.com/cnlohr/ch32v003fun/blob/5e3c3a322efa12bb4d398e0ec4adcf07fcad446f/.gitignore#L7

Note that there is one `*.bin` file that **should** be source controlled:

* https://github.com/cnlohr/ch32v003fun/blob/master/attic/factory_bootloader.bin

If `blink[_raw].bin` should not be there I can do a PR to remove them.

dwillmore, 2024-01-31:
I believe the idea was to have a .bin of the most basic starter example
available for people to use to test boards even if they couldn't get a
build environment set up.  I may be mis-remembering that.

TommyMurphyTM1234, 2024-01-31:
Thanks @dwillmore - that makes sense.
I'll close this issue as invalid so.
Thanks again.

closed 2024-01-31 by TommyMurphyTM1234

TommyMurphyTM1234, 2024-01-31:
I guess you're correct @dwillmore - these commits certainly suggest that these binary files were added deliberately and not by accident.

* https://github.com/cnlohr/ch32v003fun/commits/master/examples/blink/blink.bin
* https://github.com/cnlohr/ch32v003fun/commits/master/examples/blink_raw/blink_raw.bin

dwillmore, 2024-01-31:
Even a broken clock is right twice a day. :)

cnlohr, 2024-02-01:
It was my fault - It was when I bifrucated the blink example between blink using higher level macros / low level registers.

TommyMurphyTM1234, 2024-02-01:
> It was my fault - It was when I bifrucated the blink example between blink using higher level macros / low level registers.

Thanks @cnlohr - does that mean that they should be removed or should things be left as is? If the former I can do a PR for that small change if you want.

BTW I just noticed that, funnily enough, the "raw" version is actually 4 bytes larger than the "high level" version when one might assume that using a higher level abstraction API might involve a size penalty. Nice! 🙂

* https://github.com/cnlohr/ch32v003fun/issues/265#issuecomment-1917597036

TommyMurphyTM1234, 2024-02-21:
Hi @cnlohr - I see that you removed `blink_raw.bin` here:

* https://github.com/cnlohr/ch32v003fun/commit/2108ebeac583261a3a599e0e64d565722fb94e4d

Did you also mean to remove `blink.bin` or is that deliberately left intact as a precompiled binary example that can be used without having to build anything or set up the development environment properly?

* https://github.com/cnlohr/ch32v003fun/blob/master/examples/blink/blink.bin

dwillmore, 2024-02-21:
>    - 2108ebe

The precompiled blink.bin file is provides as a quick start for new users
trying to test out their boards.  Its inclusion is intentional.

TommyMurphyTM1234, 2024-02-21:
> The precompiled blink.bin file is provides as a quick start for new users trying to test out their boards. Its inclusion is intentional.

Thanks @dwillmore but that still doesn't tally with what @cnlohr said previously so it would be useful to see what he says:

* https://github.com/cnlohr/ch32v003fun/issues/272#issuecomment-1922129727

> It was my fault - It was when I bifrucated the blink example between blink using higher level macros / low level registers.

dwillmore, 2024-02-21:
>    - #272 (comment)

The _raw version was the oversight, the normal .bin was intentional.

dwillmore, 2024-02-21:
This has been discussed on the discord.  You could join and I can send you links to the relevant posts.