cnlohr/ch32fun #280: Configurable RAM sizes for V20x, V30x

maxgerhardt, 2024-02-18:
The linker script

https://github.com/cnlohr/ch32v003fun/blob/6c14912e79adf9097d46c3f4688da3b567d64d6c/ch32v003fun/ch32v003fun.ld#L31-L40

for V30x only gives the "256K Flash + 64K RAM" and "128K Flash + 32K RAM" option. This is however wrong: If the 128K Flash option doesn't exist naturally. The options for 30x are

[image]

And for 20x

bmx, 2024-02-18:
[image]
Who's wrong? the datasheet? the linkutility?

cnlohr, 2024-02-18:
> Through invocation of the blackest dark magic I've ever seen, flash cells become RAM cells, and their sum is a constant.

wat

Can you validate the actual performance?

maxgerhardt, 2024-02-18:
> Who's wrong? the datasheet? the linkutility?

Ah, indeed I'm wrong, or rather, confusingly worded: The V30x devices marked "256K Flash + 64K SRAM" parts are actually configurable for the 4 options shown in the link utility. This is only the CH32V307(RC/WC/VC) and CH32V303(RC/VC) chips. For the other chips, 303(CB,RB), 305(FB,RB), they **do** have 128K flash and 32K RAM *fixed*.

[image]
[image]

For V20x series chips, only V203RB has configurable sizes from selections, all others are 32K+10K or 64K+20K.

The point still stands however, V30x sreies chips can actually have 5 different configurations, the `.ld` file makes only 2 possible.

cnlohr, 2024-02-19:
@dwillmore has been postulating that the flash is in fact RAM, and that it's somewhat arbitrary how it's configured. I am extremely curious if we find something like that.   It seems that there are bits in the option rom for configuring the bits, see `FLASH_OBR`
[image]

@maxgerhardt would you be up for adding to the .ld file to enable the other forms?  Also, it seems you will need to flash the option bytes in order to select between the different modes.  Do you have any recommendations on how we could select the appropriate split?

cnlohr, 2024-02-19:
Wait... does that mean it can be configured at runtime?  I **also** see it here:

[image]

jnk0le, 2024-04-22:
> > Through invocation of the blackest dark magic I've ever seen, flash cells become RAM cells, and their sum is a constant.
> 
> wat

> @dwillmore has been postulating that the flash is in fact RAM, and that it's somewhat arbitrary how it's configured

This flash is in fact a sram loaded with image from "external" flash: [htps://zeptobars.com/en/read/GD32F103CBT6-mcm-serial-flash-Giga-Devices](https://zeptobars.com/en/read/GD32F103CBT6-mcm-serial-flash-Giga-Devices)

jnk0le, 2024-05-01:
someone already decapped it https://cpldcpu.wordpress.com/2024/05/01/decapsulating-the-ch32v203-reveals-a-separate-flash-die/