cnlohr/ch32fun #284: Make minichlink print flash amount

cnlohr, 2024-03-02:


cw2, 2024-03-22:
What do you mean by that? minichlink already prints flash amount during setup:

```
Found WCH Link
WCH Programmer is CH32V307 version 2.11
Detected: CH32V003
Setup success
Flash Storage: 16 kB
...
```
So is the issue to (e.g.) add a new command to output just the flash amount separately?

TommyMurphyTM1234, 2024-06-19:
Perhaps @cnlohr meant to print the amount of flash actually written during programming? E.g. the start address and extent or something like that?
```
../../ch32v003fun/../minichlink/minichlink -w blink.bin flash -b
Found WCH Link
WCH Programmer is LinkE version 2.11
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : ...
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
Image written.
```

cnlohr, 2025-11-25:
I believe this is an issue on the ch32v006, it displays some stupendous amount of flash available IIRC.