cnlohr/ch32fun #286: Empty LD file on missing build tools

furosha, 2024-03-03:
Hi,

This is probably going to be a gotcha for beginners, it took me a while to figure out.

Building the blink example for the CH32V003 and i was getting a weird error on the ELF:

`
~/projects/ch32v003fun/examples/blink$ make clean
rm -rf blink.elf blink.bin blink.hex blink.lst blink.map blink.hex || true

~/projects/ch32v003fun/examples/blink$ make
riscv64-unknown-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -L../../ch32v003fun/../misc -lgcc -T ../../ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
riscv64-unknown-elf-size blink.elf
   text	   data	    bss	    dec	    hex	filename
      0	      0	      0	      0	      0	blink.elf
riscv64-unknown-elf-objdump -S blink.elf > blink.lst
riscv64-unknown-elf-objdump -t blink.elf > blink.map
riscv64-unknown-elf-objcopy -O binary blink.elf blink.bin
riscv64-unknown-elf-objcopy: error: the input file 'blink.elf' has no sections
make: *** [../../ch32v003fun/ch32v003fun.mk:146: blink.bin] Error 1
`

The source turned out to be an empty generated_ch32v003.ld:
`
total 788
drwxrwxr-x  2 _________ _________   4096 Mar  3 14:49 ./
drwxrwxr-x 14 _________ _________   4096 Mar  3 13:17 ../
-rw-rw-r--  1 _________ _________   3604 Mar  3 13:10 ch32v003fun-bootloader.ld
-rw-rw-r--  1 _________ _________  56470 Mar  3 13:10 ch32v003fun.c
-rw-rw-r--  1 _________ _________ 720100 Mar  3 13:10 ch32v003fun.h
-rw-rw-r--  1 _________ _________   4658 Mar  3 13:10 ch32v003fun.ld
-rw-rw-r--  1 _________ _________   5130 Mar  3 14:49 ch32v003fun.mk
-rw-rw-r--  1 _________ _________      0 Mar  3 13:10 generated_ch32v003.ld`

I think I caused this by running build before installing the toolchain.

There are probably a few solutions to this, but I dont think it would hurt to check the generated LD is not empty in the build scripts.
That seems like a reasonable sanity check.

Kind Regards.

BogdanTheGeek, 2024-03-04:
same as #283

furosha, 2024-03-04:
Weird that did not come up when I searched.
This can be closed as a duplicate.

closed 2024-03-04 by furosha