cnlohr/ch32fun #290: prebuilt blink.bin example doesn't seem to work on a CH32V003 board

brian-smith-github, 2024-03-08:
Hi there. 
When getting started with the CH32V003 board I first tried flashing the provided examples/blink/blink.bin file, but nothing happened.

When I re-built it from source, the new binary is a different size - 560 bytes vs 544 of the previous version, and this one worked fine.

cnlohr, 2024-03-09:
Weird - I am going to leave this open for now, in case someone else sees what's going on.

TommyMurphyTM1234, 2024-06-14:
> When getting started with the CH32V003 board I first tried flashing the provided examples/blink/blink.bin file, but nothing happened.

I can't reproduce this problem.
I did this (on Windows, hence the use of the backslash directory separator)...
```
git clone https://github.com/cnlohr/ch32v003fun
cd ch32v003fun\examples\blink

# Erase flash
..\..\minichlink\minichlink.exe -E
Found WCH Link
WCH Programmer is LinkE version 2.11
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : ...
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
Whole-chip erase

# Program pre-built blink.bin
..\..\minichlink\minichlink.exe -w blink.bin flash -b
Found WCH Link
WCH Programmer is LinkE version 2.11
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : ...
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
Image written.
```
... and LED1 started blinking as expected.

> When I re-built it from source, the new binary is a different size - 560 bytes vs 544 of the previous version

This is probably not surprising unless the exact same toolchain and options are used to compile the `blink` example from sources as were used to compile the pre-built `examples/blink/blink.bin` binary.

For example, I downloaded and extracted the xPack RISC-V Embedded GCC:

- https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  - https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v13.2.0-2/
    - https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/download/v13.2.0-2/xpack-riscv-none-elf-gcc-13.2.0-2-win32-x64.zip

and also the xPack Windows Build Tools:

- https://xpack.github.io/dev-tools/windows-build-tools/
  - https://xpack.github.io/dev-tools/windows-build-tools/releases/
    - https://github.com/xpack-dev-tools/windows-build-tools-xpack/releases/tag/v4.4.1-2/

(and I copied the xPack Windows Build Tools binaries into the toolchain `bin` directory for convenience)

and then built the `blink` example from sources:
```
PATH C:\temp\xpack-riscv-none-elf-gcc-13.2.0-2\bin;%PATH%

# Backup the pre-built binary
cp blink.bin blink.bin.bak

make clean
rm -rf blink.elf blink.bin blink.hex blink.lst blink.map blink.hex ../../ch32v003fun/generated_ch32v003.ld || true

make PREFIX=riscv-none-elf
riscv-none-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 ../../ch32v003fun/ch32v003fun.ld > ../../ch32v003fun/generated_ch32v003.ld
riscv-none-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -L../../ch32v003fun/../misc -lgcc -T ../../ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
riscv-none-elf-size blink.elf
   text    data     bss     dec     hex filename
    564       0       0     564     234 blink.elf
riscv-none-elf-objdump -S blink.elf > blink.lst
riscv-none-elf-objdump -t blink.elf > blink.map
riscv-none-elf-objcopy -O binary blink.elf blink.bin
riscv-none-elf-objcopy -O ihex blink.elf blink.hex
make -C ../../ch32v003fun/../minichlink all
make[1]: Entering directory 'C:/temp/ch32v003-fun/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory 'C:/temp/ch32v003-fun/ch32v003fun/minichlink'
../../ch32v003fun/../minichlink/minichlink -w blink.bin flash -b
Found WCH Link
WCH Programmer is LinkE version 2.11
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : ...
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
Image written.

dir blink.bin*
 Volume in drive C has no label.
 Volume Serial Number is 06A0-A88B

 Directory of C:\temp\ch32v003-fun\ch32v003fun\examples\blink

14/06/2024  17:38               564 blink.bin
14/06/2024  16:58               560 blink.bin.bak
               2 File(s)          1,124 bytes
               0 Dir(s)  255,119,859,712 bytes free
```
```

brian-smith-github, 2024-06-14:
Hi there.
Please note that the blink.bin was replaced on April 3, this issue as raied on March 8, so I'd assumed it had already been fixed. I will close this issue to avoid further confusion!

closed 2024-06-14 by brian-smith-github

TommyMurphyTM1234, 2024-06-14:
> Hi there. Please note that the blink.bin was replaced on April 3, this issue as raied on March 8, so I'd assumed it had already been fixed. I will close this issue to avoid further confusion!

Ah, OK - I didn't realise that the issue predated a subsequent change to the binary! Thanks for the clarification.