cnlohr/ch32fun #292: PWM 8 KHz 16 bit

AAJAY5, 2024-03-12:
Any suggestions for PWM 8 KHz 16 bit

dwillmore, 2024-03-12:
What do you mean suggestions? Yes, it's a thing.

AAJAY5, 2024-03-12:
Actually I was able to play 8 bit 8 KHz wav ( PCM ) using PWM but I couldn't find any working solution to get 8 KHz 16 bit PWM output for the same. Is that possible?

dwillmore, 2024-03-12:
You should be able to get close.  24MHz doesn't have an exact divider to get 8KHz * 16 bits, but it can get close.  Divide the 24MHz clock by 187 or 188 and you'll get close.  The exact divisor is 187.5

AAJAY5, 2024-03-12:
As per document

Freq : 48 MHz
Pres: 1
PWM Freq : 8 Khz
Max PWM levels ~5952 which is 12 bit ( 4096 )

AAJAY5, 2024-03-12:
> You should be able to get close.  24MHz doesn't have an exact divider to get 8KHz * 16 bits, but it can get close.  Divide the 24MHz clock by 187 or 188 and you'll get close.  The exact divisor is 187.5
> 

Can you please send quick snippet?

AAJAY5, 2024-03-12:
> I'm not sure what document you're refering to.  If you're using the 48MHz clock, then a divisor of 375 will give you exactly 8KHz at 16 bits.
> 

Confused where to set divisor 375?

dwillmore, 2024-03-12:
That would be the timer prescaler value.  Check the Reference Manual chapter on the TIMER peripherials.

dwillmore, 2024-03-12:
There is example timer code in this very repo.  Please familiarize yourself with it.

cnlohr, 2024-03-19:
Going to close this for now due to inactivity.