cnlohr/ch32fun #294: CH32V003A4M6 SPI?

bt103, 2024-03-22:
I'd prefer to use the chips with the wider pitch SOP package, however the manufacturer datasheet shows that neither of them have SPI.  However when I look at the picture of the CH32V003A4M6 on this github page, it shows SPI pins.  Does the CH32V003A4M6 indeed have SPI?

AlexanderMandera, 2024-03-22:
No, the hardware SPI is on pins PC5-PC7.
It is not available on the 8-pin version, only PC1 which is SPI NSS is exposed.
But it doesn't help on its own.

bt103, 2024-03-22:
> No, the hardware SPI is on pins PC5-PC7. It is not available on the 8-pin version, only PC1 which is SPI NSS is exposed. But it doesn't help on its own.

Thanks, I think I see my error now.  When I looked at the CH32V003A4M6 picture, I saw MOSI and MISO and thought SPI was there.  But the clock signal is missing.  Bummer :-(

closed 2024-03-22 by cnlohr

maroziza, 2024-08-17:
I Have a thory looking through the manual, but not sure if it will workout. PC5 is almost certainly is on die, and it is shared with Timer. if we passthrught PC5 clock signal using timer to different pin,  we can use hardware spi losing 2 PWM channels.
PC5 can be: TIM1_ETR, TIM1_CH3, TIM2_ETR, TIM2_CH1
So we can put timer in trigger mode, just copying CLK pulse as fast as possible. 
It is very interesting for me to try, but i don't have chips on my hand.