cnlohr/ch32fun #295: Is this board right for me?

LimesKey, 2024-03-26:
Hi, I'm planning on using `CH32V003J4M6` for an upcoming PCB hardware project of mine but I'm not sure if this is viable for what I want it to do. I couldn't find a better place to ask a question to those experienced with this board specifically so I'm posting here.

My use case for this microcontroller is to light up a couple of LEDs for 10 minutes when a specific pin receives power. I'm using this board as a simple battery power indicator for a LIR2032 battery which has a `DW01A` battery management system controller with OC (overcharge) and OD (over-discharge) pins connected to a MOSFET.
[image]

When the OC or OD pin is set to "low" (mosfet triggered, no power to pin) I would like it to turn on a specific LED for a time delay depending on if it was caused by either OC or OD. 
[image]

Please let me know if you'd like me to clarify anyone, I'm only a novice in PCB design and RISC-V microcontrollers.

prosper00, 2024-03-27:
seems pretty straightforward, really any microcontroller ought to handle that.

Jumping straight into ch32v as a complete newb might be a bit of a steep learning curve though.

LimesKey, 2024-03-27:
I'm creating a PCB for an upcoming event so I need to know if it works before my deadline. I've ordered all the necessary parts from AliExpress but their shipping takes at least 2 - 5 weeks and ordering the PCB takes a week. Would someone be able to confirm this would work with the flasher and code?

From AliExpress, I've ordered the `WCH-LinkE Mini` and a few ESP32s. But I'm now planning on integrating a USB-C port right into the PCB to flash the `CH32V003J4M6` module using this [guide](https://github.com/cnlohr/rv003usb). I've also been hoping I can just use the regular Arduino IDE and write code similarly to other Arduino projects, I'm planning on using [this ](https://github.com/AlexanderMandera/arduino-wch32v003)for that.

Photos below are of my new design
[image]
[image]

cnlohr, 2024-03-27:
This looks weeeird. 
[image]

Also, I've not used the SOIC-8 version with USB so I just don't know. 

Regardless, it's going to be very hard to get everything right on the first go.

LimesKey, 2024-03-28:
I fixed that right after I sent the message. Well thanks though

vegetablesalad, 2024-04-07:
Honestly just using some 555 latches or any other "analog" approach might be way easier here. Especially if this is a one time project you need to get it up running quickly.

closed 2024-08-23 by LimesKey