cnlohr/ch32fun #299: Are there defined symbolic constants for AFIO_EXTICR register values?

IvanLi-CN, 2024-04-05:
I tried using the following set of definitions, but found that the values were different from those in the manual.
 https://github.com/cnlohr/ch32v003fun/blob/7b87b3fc33d96cf10b934985e13ce63602dd8457/ch32v003fun/ch32v003fun.h#L4737-L4773

Here's a snippet of my code.

```c
  // Enable interruptions for PC0, PC1
  AFIO->EXTICR |= AFIO_EXTICR1_EXTI0_PC | AFIO_EXTICR1_EXTI1_PC;

  // Enable interruptions
  EXTI->INTENR |= EXTI_INTENR_MR0 | EXTI_INTENR_MR1;
  EXTI->FTENR |= EXTI_FTENR_TR0 | EXTI_FTENR_TR1;
```

Below is a screenshot of the manual.

[image]

I see that [the example](https://github.com/cnlohr/ch32v003fun/blob/7b87b3fc33d96cf10b934985e13ce63602dd8457/examples/exti_pin_change_isr/exti_pin_change_isr.c#L87C2-L87C27) is directly assigned, but I found a similar defined, but the value does not meet my expectations. I don't know if there is a problem with my usage, please ask for some help, thank you for this project and you who see this issue.

cnlohr, 2024-04-28:
I believe this has been corrected at another time, I don't recall the specific commit.  Can you check again?

IvanLi-CN, 2024-04-28:
@cnlohr Thanks for the reply. I just rechecked the code in the master branch and the problem persists. I compared the documentation for CH32V203 and it seems that CH32V003 is completely different from the others in this section.

https://github.com/cnlohr/ch32v003fun/blob/8c421623720a64b631d78aa4255abac702fdf40a/ch32v003fun/ch32v003fun.h#L5262-L5301

[image]

prosper00, 2024-04-28:
> [Tracking image][https://github.com/notifications/beacon/ANVXEPP5FWPZPBK55NVL6HTY7TQ6JA5CNFSM6AAAAABFY4X3O2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT4CCFCE.gif]

i ran into the same thing. will be opening up a PR in the next couple of days. In the meantime, you can copy/paste the defines from 'funconfig.h' in my repo here:

https://github.com/prosper00/Encoder-i2c

cnlohr, 2024-04-28:
> i ran into the same thing. will be opening up a PR in the next couple of days. In the meantime, you can copy/paste the defines from 'funconfig.h' in my repo here: https://github.com/prosper00/Encoder-i2c Apr 28, 2024 6:07:00 a.m. Ivan Li ***@***.***>:

Would you like me to make a PR on your behalf, it would be good to fix this.

prosper00, 2024-04-28:
i opened on this morning: [cnlohr/ch32v003fun] Pull request opened: #309 Exti defs

closed 2024-05-02 by IvanLi-CN