cnlohr/ch32fun #301: DISCUSSION - Could this run Klipper ? I think it really could omg

shodanx2, 2024-04-08:
Hi,

Klipper is probably 3D printing's most advanced firmware.

It is a two part system where the hard math is done on a real computer, and the real time stuff is done on a microcontroller.

A solution the LinuxCNC project is still in denial about, but I digress.

As a result, the MCU portion of klipper has relatively few functions to implement

https://www.klipper3d.org/MCU_Commands.html

Also one thing to note about Klipper, it can use multiple MCUs at one.

So, instead of one board that has everything, you could have one MCU only do X/Y axis, have another do the Z(1,2,3,4) plus the fans, rgb and bed heater and yet another only do the extruder motor and heater

That means the PCB could be much much simpler than currently monolithic approach

However, nobody in the 3dprinting community has heard of the CH32V003, so there are no such boards yet.

3D printer board are not super complicated. Here are a few of the most popular open source design, based on the MKS Skipr design I believe.
[
BTT SKR 1.4](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf)

[BTT SKR V2](https://github.com/bigtreetech/SKR-2/blob/master/Hardware/BIGTREETECH%20SKR%202-SCH.pdf)

[BTT SKR V3](https://github.com/bigtreetech/SKR-3/blob/master/Hardware%20(SKR%203)/BIGTREETECH%20SKR%203-SCH.pdf)

[BTT SKR mini E3V3](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20E3%20SKR%20MINI%20V3.0_PIN.pdf)

found the SKR Skipr..  I might have been wrong

[MKS Skipr](https://github.com/makerbase-mks/MKS-SKIPR/blob/main/hardware/MKS%20SKIPR%20V1.0_002/MKS%20SKIPR%20V1.0_002%20SCH.pdf)

But wait there's more !

The Klipper host and the Klipper MCU, communicate with one another using either serial port or canbus

But I believe it could be using ethernet ALMOST transparently without modifying the Klipper host software, only some ethernet to serial shenanigans in the MCU firmware. 

Using the one of ENC28J60,LAN8720 or W5500

I asked about it on the klipper forum and got 0 replies !

https://klipper.discourse.group/t/can-i-add-ethernet-support-to-my-klipper-stm32f103-enc28j60-lan8720-or-w5500/14733

Which I take as a lack opposition, win !

So there already are many firmware do ethernet to serial bridge 

https://github.com/oxan/esphome-stream-server

https://github.com/yuri-rage/ESP-Serial-Bridge
https://github.com/AlphaLima/ESP32-Serial-Bridge
https://hackaday.io/project/161491-qmesh-a-lora-based-voice-mesh-network/log/194667-esp32-serial-to-wifi-bridge-complete

oops I have better links for that , but 
My computer is about to crash from having  this many

[image]

tabs open, I will now post and come back after the reboot

shodanx2, 2024-04-16:
So quick survey, 

I found that Klipper, the MCU component of klipper is compatible with old 8mhz arduinos

https://github.com/Klipper3d/klipper/tree/2425a74638baa87efef3ca02253804d126101c8a/src/avr

I'm not sure, but I think that folder, is the whole thing, as far as the MCU firmware is concerned .... I might be able to port it with some help from chatgpt ....

As for ethernet 

The 3 chips ENC28J60,LAN8720 or W5500 that are most in favour cost about 3-4$ , 2-3$, 4-5$ respectively. 

These are the libraries, I have no idea how hard it would be to make those work on the CH32V003, perhaps it works out of the box ?

https://github.com/ARM-software/CMSIS-Driver

https://github.com/UIPEthernet/UIPEthernet

https://github.com/afiskon/stm32-w5500

shodanx2, 2024-04-16:
Explanation of each chip by chatgpt

Certainly! The ENC28J60, LAN8720, and W5500 are Ethernet chips commonly used in embedded systems and IoT devices to enable network connectivity. Here’s a breakdown of each:
1. ENC28J60

The ENC28J60 from Microchip Technology is a standalone Ethernet controller with an industry standard Serial Peripheral Interface (SPI). It is designed to serve even low-cost, low-speed, less-complex microcontrollers with Ethernet capabilities.

Key Features:

    Interface: SPI, which makes it simple to interface with most microcontrollers.
    Speed: Supports 10 Mbps Ethernet.
    Package: Available in multiple packages including PDIP, QFN, and SOIC, making it suitable for hobbyists and professional designs.
    Stack: It requires implementing or having a TCP/IP stack on the microcontroller, as it primarily handles only the physical and data link layers.

2. LAN8720

The LAN8720 is a low-power 10BASE-T/100BASE-TX physical layer (PHY) transceiver from Microchip (formerly from SMSC). It interfaces with microcontrollers via the Media Independent Interface (MII) or Reduced Media Independent Interface (RMII).

Key Features:

    Interface: MII/RMII interfaces which are common in many microprocessors and microcontrollers.
    Speed: Supports both 10 Mbps and 100 Mbps Ethernet speeds, offering more flexibility than ENC28J60.
    Power Efficiency: Extremely low power consumption, which is ideal for power-sensitive applications.
    Applications: Suitable for applications where integration with an MCU with an existing MAC layer is needed.

3. W5500

The W5500 from WIZnet is a hardwired TCP/IP embedded Ethernet controller that enables easier Internet connection for embedded systems using SPI.

Key Features:

    Interface: SPI for interfacing with the host controller.
    Integrated TCP/IP stack: Unlike ENC28J60, it includes a fully hardwired TCP/IP stack, which can offload significant processing tasks from the main microcontroller, leading to robust and high-speed network communications.
    Speed: Supports 10/100 Mbps Ethernet.
    Socket Support: Provides up to 8 simultaneous socket connections, making it suitable for complex networked applications.

Comparison Summary:

    ENC28J60 is a good choice if cost and simplicity are crucial, and only 10 Mbps speed is sufficient.
    LAN8720 serves better in applications needing higher throughput (100 Mbps) and where a PHY layer with a flexible interface (MII/RMII) to a MAC layer in a microcontroller is needed.
    W5500 excels in applications where ease of implementation of network functionality with minimal load on the host processor is desired, thanks to its integrated TCP/IP stack and multi-socket capability.

Each of these chips is targeted at different network scenarios, so the choice depends largely on the specific needs and constraints of your project, such as speed requirements, power consumption, and available microcontroller interfaces.

shodanx2, 2024-04-16:
Pictures of the ENC28J60, off the shelf boards

[image]
[image]
[image]
[image]

Picture of the LAN8720 
[image]
[image]

Pictures of the W5500
[image]
[image]

shodanx2, 2024-04-18:
For future reference, TI has an open source (but license unknown) ethernet to serial firmware, based on lwIP and FreeRTOS

https://www.ti.com/lit/an/spma072/spma072.pdf?ts=1713345916020\
http://www.ti.com/lit/zip/spma072

"this software for use solely and
exclusively on TI's microcontroller products"

"You may not combine this software with "viral" open-source
software in order to form a larger program."

So, that will be as a source of inspiration but no copying allowed !

cnlohr, 2024-04-22:
Would you like to join my Discord, to discuss?  We have better discussions there than on Discord.  You can ping me, my username is `cnlohr`

Everything I've head about klipper is that it's very heavy weight.  Could a CH32V003 do full CNC control with tons of features over ethernet, sure! But, probably not with klipper would be my guess.

cnlohr, 2024-04-22:
Also, side-note:  You know, there are CH32XXX's with ethernet built in, right?  Check out the ch32v307

shodanx2, 2024-04-23:
Thank you for the invitation, I might stop by discord, but I really do not like discord and what it has done to public knowledge on the internet and do not have an account. 

I am far more enticed by your video abour VR chat as a kind of internet underground and might meet your there if you still go !

I think you are correct, klipper might be too big for the CH32V003.

I have given this a try, to simply compile klipper for the atmega328P, a MCU with 32K (as you know, double that of the CH32V003)

They would use this kind of board perhaps 

[image]

I have documented my attempt, and even a mutilated version of klipper, I could not get below 37K and I think that doesn't even include a bootloader (which might be another 4K)

https://github.com/gwisp2/klipper-drawbot/issues/1#issuecomment-2071565458

Here is the actual code

https://github.com/Klipper3d/klipper/tree/master/src/avr

I used [kiauh](https://github.com/dw-0/kiauh) on debian 12 to set this up but I had to roll back gcc avr to debian 11 to build at all. 

However,   

I think I might have gotten something wrong

There people claim that with only the support for the accelerometer they got it down to 2800 bytes 

https://klipper.discourse.group/t/cannot-build-firmware-for-arduino-uno/7041/7

Perhaps there is hope.

I wonder how much space the ethernet would need. 

I didn't know about the CH32V307RCT6, they are 2.58$CAD in 100qty on ali
[image]

The package has many more pins however. 10x10mm LQFP-64M

That is certainly an option to consider
..... oh, it has 8 USARTS .... and two canbus

[image]

That's a lot of stuff for 2.58$CAD

I hope the CH32V003 is still somehow viable ... but wow there is a lot of room to grow here !

cnlohr, 2024-04-23:
After chatting with some folks from my server, I think it is viable, but, what's more important than spending time figuring out what processor to use for a project is starting to do projects with a processor and seeing where you land.

If you get stuck, someone can probably help out.

And Discord is good for a more private thing.  We do the best we can to disseminate information when it's prepared for release.

rjjrbatarao, 2024-08-06:
This is a great idea cheap way to isolate controls on a 3d printer. chv chips are amazing i've already done few project with them with reliable performance. My 3d printers are collecting dust right now this might give me some fuel again to start 3d printing again.

shodanx2, 2024-08-06:
Hi,

For now I am investigating using a dedicated serial to ethernet bridge, that is the CH9121 chip. ali item 1005007492381945

The cheapest I could find was 5.32$CAD/each for 10x, so that is already 17 times more expensive than the ch32v003 and you still need a mcu to run klipper proper.

The ch32v003 can't run a whole printer on its own, but multiple of them probably could.  I would imagine something like, one for the extruder control (1 stepper plus 1 PID controlled heater) and one for X/Y/Z+bed. 

If these two chips (and many a 3rd one for accessories) could talk some 1mbps ethernet and hook into a nearby switch, that could make managing a whole shelf of printers really convenient and it would work with bottom of the barrel printers.

closed 2024-10-21 by eeucalyptus