cnlohr/ch32fun #303: v30x blink example don't run if TARGET_MCU_PACKAGE sets to CH32V307VCT6

romainreignier, 2024-04-13:
Hi,
First, I would like to thank you for this amazing project.
I have successfully tried the blink example from the `examples_v30x` on a `CH32V307 EVT by SCDZ` board.
But the board is actually populated with a `CH32V307VCT6` and the Makefile selects a CH32V305RBT6, I have then replaced it with the following:

```
all : flash

TARGET:=blink
TARGET_MCU:=CH32V307
TARGET_MCU_PACKAGE:=CH32V307VCT6

include ../../ch32v003fun/ch32v003fun.mk

flash : cv_flash
clean : cv_clean
```

And after a `make clean` and then `make`, the code does not run once uploaded to the board.

I have also tried with the `debugprintfdemo` copied from the V003 examples, it works with the V305RBT6 settings but not with the V307VCT6.

From the `ch32v003fun.mk`, the difference is only the `MCU_PACKAGE` and the sizes for the FLASH and RAM.

I have then tried to configure `TARGET_MCU_PACKAGE:=CH32V307RBT6` and in that case, the code runs.

When it does not work, I have tried to use GDB and adding a breakpoint on the first line of the main on the `SystemInit()` line. The breakpoint is reached but on first step, I am in the `DefaultIRQHandler`.

Does anyone have a clue of why I have this behavior when using the config matching my MCU?

cnlohr, 2024-04-22:
Sadly, at the moment no one is looking into this specific target.  If you are up for seeing where we've made a mistake, it would be great to let us know. 

There's plarform-specific code in the .ld, Makefile, and ch32v003.h

As a note, when changing the .ld, you will need to remove the auto-generated .ld for your part in the ch32v003fun folder.

Please let us know what you find.

cnlohr, 2024-04-24:
@AlexanderMandera - do you recall if you ever tested with CH32V307VCxx?  Should this line read 2 instead of 1? 			MCU_PACKAGE:=1 https://github.com/cnlohr/ch32v003fun/blob/6c3c515598db6d3d50f8087da2f694ee417fc445/ch32v003fun/ch32v003fun.mk#L114C14-L114C18

@romainreignier Can you test with changing that line of that file from 2 to 1, removing the generated .ld file, and seeing if it works then?

romainreignier, 2024-04-26:
If I change the 1 to a 2, it works but the linker script now contains 128K of flash and 32K of ram while according to the datasheet, the VC variant should be 256/64:
[image]

cnlohr, 2024-04-27:
I am just not familiar with the 30x line, does it make sense to add another package type for the 307?  You can copy-paste it from one of the other parts, and make a PR.

bmx, 2024-05-11:
`make TARGET_MCU=CH32V307 TARGET_MCU_PACKAGE=CH32V307VCT6`

it should work

```
MEMORY
{
  FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 256K
  RAM (xrw) : ORIGIN = 0x20000000, LENGTH = 64K
}
```

monte-monte, 2025-11-25:
I believe it should work now.